rusefi/rusefi #4: better/faster idle control logic

rusefi, 2017-02-21:
We need proper PID and maybe more in order to get usable idle valve control.

This replaces https://sourceforge.net/p/rusefi/tickets/216/

rusefi, 2017-02-21:
See also https://github.com/rusefi/rusefi/issues/209

rusefi, 2017-05-11:
Also need some defection of idle mode - open question

See also http://rusefi.com/forum/viewtopic.php?f=2&t=1230&p=25865#p25775

stefanst, 2017-05-11:
CL idle ideally needs the following (at least)

Idle rpm target curve (target rpm over CLT)

Intelligence to recognize if CL idle should be engaged.
- i.E. no CL idle if TPS is open (minimum percentage for analog TPS or TP-Switch)
- Some way of recognizing that engine is being driven by wheels (easiest would be clutch switch and or neutral switch on manuals)
- Max rpm for entering CL idle (should be target idle rpm + maybe 200 or so)

Initial value table: 2D table showing initial idle PWM based on target rpm and CLT. Could just be 1D, based on CLT if need be, but 2D would be more flexible

Awesome addition would be a few conditional adders to PWM: PWM needs to increase for example when AC kicks in, automatic transmission is engaged, on smaller engines even the power-steering-pump engaging (driver turning the steering wheel while standing) can cause a significant drop. Also cooling fan coming on may require increase in PWM.

rusefi, 2017-05-17:
@stefanst  when you say "Initial value table" is it what we call PID offset? Are you suggesting offset curve instead of one fixed PID offset?

stefanst, 2017-05-17:
@rusefi That is correct. Different idle rpms require different PWMs at different CLTs. A 2D table (CLT on the x-axis, RPM on the y-axis and PWM percentage offset in the cells) will take care of that.
Additional offsets can come in for additional predictable loads like radiator fan and AC.

radekMX5, 2017-10-18:
I don't know much about this but I try to help anyway :) Maybe this can be useful? 
http://www.megamanual.com/ms2/IAC.htm

mck1117, 2020-07-28:
this is ancient, we can close

rusefillc, 2020-07-29:
@andreika-git ?

andreika-git, 2020-07-29:
we can close

closed 2020-07-29 by rusefillc