rusefi/rusefi #375: why are we using softfp and not hard FPU?

rusefi, 2017-03-20:
mobyfab is wondering why is rusEfi currently using softfp and not hard FPU support

See http://rusefi.com/forum/viewtopic.php?f=5&t=373&sid=a7351f5bc429bc2a5a6e2a8bf5a041b0&p=24447#p24447

rusefi, 2017-03-20:
See also https://github.com/rusefi/rusefi/issues/72

rusefi, 2017-03-20:
softfp:
set engine_type 0
enable self_stimulation

2017-03-20 17_53: EngineState: trigger error extra LED: NONE OM_DEFAULT
2017-03-20 17_53: EngineState: primary logic input: NONE
2017-03-20 17_53: EngineState: secondary logic input: NONE
2017-03-20 17_53: EngineState: zeroTestTime=14 maxHowFarOff=5725
2017-03-20 17_53: EngineState: advanceLookupTime=1152 now=416718634 fuelCalcTime=4062
2017-03-20 17_53: EngineState: ignitionSchTime=0 injectonSchTime=0
2017-03-20 17_53: EngineState: mapTime=0/hipTime=0/rpmTime=0/mainTriggerCallbackTime=0
2017-03-20 17_53: EngineState: maxLockTime=0 / maxTriggerReentraint=0
2017-03-20 17_53: EngineState: maxEventQueueTime=2614
2017-03-20 17_53: EngineState: hipLastExecutionCount=0
2017-03-20 17_53: EngineState: hwSetTimerTime 451
2017-03-20 17_53: EngineState: totalTriggerHandlerMaxTime=0

rusefi, 2017-03-20:
eclipse version does not compile:
Building file: ../kill_for_coverity.c
Invoking: Cross ARM C Compiler
arm-none-eabi-gcc -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -O0 -fmessage-length=0 -ffunction-sections -fdata-sections -flto  -g3 -DCORTEX_USE_FPU=TRUE -I".." -I"../hw_layer/stm32f4" -I../chibios/os/kernel/include -I"../ext" -I"../ext_algo" -I"../chibios/os/various" -I"../hw_layer/algo" -I"../util" -I"../console" -I"../console/binary" -I"../config" -I"../config/engines" -I"../controllers" -I"../controllers/algo" -I"../controllers/core" -I"../controllers/trigger" -I"../controllers/sensors" -I"../controllers/system" -I"../controllers/math" -I"../development" -I"../development/test" -I"../development/hw_layer" -I"../hw_layer" -I"../hw_layer/lcd" -I"../hw_layer/serial_over_usb" -I"../config/system" -I"../config/stm32f4ems" -I"../chibios/os/kernel/include" -I"../chibios/boards/ST_STM32F4_DISCOVERY" -I"../chibios/os/hal/src" -I"../chibios/os/hal/include" -I"../chibios/os/hal/platforms/STM32/TIMv1" -I"../chibios/os/hal/platforms/STM32/I2Cv1" -I"../chibios/os/hal/platforms/STM32" -I"../chibios/os/hal/platforms/STM32/RTCv2" -I"../chibios/os/hal/platforms/STM32/OTGv1" -I"../chibios/os/hal/platforms/STM32/GPIOv2" -I"../chibios/os/hal/platforms/STM32F4xx" -I"../chibios/os/hal/platforms/STM32/SPIv1" -I"../chibios/os/hal/platforms/STM32/USARTv1" -I"../chibios/os/ports/GCC/ARMCMx" -I"../chibios/os/ports/GCC/ARMCMx/STM32F4xx" -I"../chibios/os/ports/common/ARMCMx" -I"../chibios/os/ports/common/ARMCMx/CMSIS/include" -std=gnu11 -fgnu89-inline -c -Werror=type-limits -Werror=uninitialized -Werror=return-type -Werror=unused-but-set-variable -MMD -MP -MF"kill_for_coverity.d" -MT"kill_for_coverity.o" -c -o "kill_for_coverity.o" "../kill_for_coverity.c"
../kill_for_coverity.c: In function 'special_abort':
../kill_for_coverity.c:10:4: warning: implicit declaration of function '__coverity_panic__' [-Wimplicit-function-declaration]
    __coverity_panic__();
    ^
Finished building: ../kill_for_coverity.c
 
Building file: ../main.cpp
Invoking: Cross ARM C++ Compiler
arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -O0 -fmessage-length=0 -ffunction-sections -fdata-sections -flto  -g3 -DCORTEX_USE_FPU=TRUE -I".." -I"../hw_layer/stm32f4" -I"../chibios/os/kernel/include" -I"../ext" -I"../ext_algo" -I"../chibios/os/various" -I"../hw_layer/algo" -I"../util" -I"../console" -I"../console/binary" -I"../console/fl_binary" -I"../config" -I"../config/engines" -I"../controllers" -I"../controllers/algo" -I"../controllers/core" -I"../controllers/trigger" -I"../controllers/trigger/decoders" -I"../controllers/sensors" -I"../controllers/system" -I"../controllers/math" -I"../development" -I"../development/test" -I"../development/hw_layer" -I"../hw_layer" -I"../hw_layer/sensors" -I"../hw_layer/lcd" -I"../hw_layer/mass_storage" -I"../hw_layer/serial_over_usb" -I"../config/system" -I"../config/stm32f4ems" -I"../chibios/os/kernel/include" -I"../chibios/boards/ST_STM32F4_DISCOVERY" -I"../chibios/os/hal/src" -I"../chibios/os/various/devices_lib/accel" -I"../chibios/os/hal/include" -I"../chibios/os/hal/platforms/STM32/TIMv1" -I"../chibios/os/hal/platforms/STM32/I2Cv1" -I"../chibios/os/hal/platforms/STM32" -I"../chibios/os/hal/platforms/STM32/RTCv2" -I"../chibios/os/hal/platforms/STM32/OTGv1" -I"../chibios/os/hal/platforms/STM32/GPIOv2" -I"../chibios/os/hal/platforms/STM32F4xx" -I"../chibios/os/hal/platforms/STM32/SPIv1" -I"../chibios/os/hal/platforms/STM32/USARTv1" -I"../chibios/os/ports/GCC/ARMCMx" -I"../chibios/os/ports/GCC/ARMCMx/STM32F4xx" -I"../chibios/os/ports/common/ARMCMx" -I../chibios/os/ports/common/ARMCMx/CMSIS/include -std=gnu++11 -fabi-version=0 -fno-exceptions -fno-rtti -fgnu89-inline -c -Werror=type-limits -Werror=uninitialized -Werror=return-type -Werror=unused-but-set-variable -MMD -MP -MF"main.d" -MT"main.o" -c -o "main.o" "../main.cpp"
cc1plus.exe: warning: command line option '-fgnu89-inline' is valid for C/ObjC but not for C++ [enabled by default]
Finished building: ../main.cpp
 
Building file: ../rusefi.cpp
Invoking: Cross ARM C++ Compiler
arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -O0 -fmessage-length=0 -ffunction-sections -fdata-sections -flto  -g3 -DCORTEX_USE_FPU=TRUE -I".." -I"../hw_layer/stm32f4" -I"../chibios/os/kernel/include" -I"../ext" -I"../ext_algo" -I"../chibios/os/various" -I"../hw_layer/algo" -I"../util" -I"../console" -I"../console/binary" -I"../console/fl_binary" -I"../config" -I"../config/engines" -I"../controllers" -I"../controllers/algo" -I"../controllers/core" -I"../controllers/trigger" -I"../controllers/trigger/decoders" -I"../controllers/sensors" -I"../controllers/system" -I"../controllers/math" -I"../development" -I"../development/test" -I"../development/hw_layer" -I"../hw_layer" -I"../hw_layer/sensors" -I"../hw_layer/lcd" -I"../hw_layer/mass_storage" -I"../hw_layer/serial_over_usb" -I"../config/system" -I"../config/stm32f4ems" -I"../chibios/os/kernel/include" -I"../chibios/boards/ST_STM32F4_DISCOVERY" -I"../chibios/os/hal/src" -I"../chibios/os/various/devices_lib/accel" -I"../chibios/os/hal/include" -I"../chibios/os/hal/platforms/STM32/TIMv1" -I"../chibios/os/hal/platforms/STM32/I2Cv1" -I"../chibios/os/hal/platforms/STM32" -I"../chibios/os/hal/platforms/STM32/RTCv2" -I"../chibios/os/hal/platforms/STM32/OTGv1" -I"../chibios/os/hal/platforms/STM32/GPIOv2" -I"../chibios/os/hal/platforms/STM32F4xx" -I"../chibios/os/hal/platforms/S [... cut ...] ster arguments, rusefi.elf does not
c:/progra~1/sourcery_codebench_lite_for_arm_eabi/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld.exe: failed to merge target specific data of file C:\Users\user\AppData\Local\Temp\cctG71Uj.ltrans11.ltrans.o
c:/progra~1/sourcery_codebench_lite_for_arm_eabi/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld.exe: error: C:\Users\user\AppData\Local\Temp\cctG71Uj.ltrans12.ltrans.o uses VFP register arguments, rusefi.elf does not

rusefi, 2017-03-20:
https://github.com/elua/elua/issues/54 blames Mentor/CodeSourcery G++ Lite suggest  gcc-arm-embedded toolchain

rusefi, 2017-03-20:
Building target: rusefi.elf
Invoking: Cross ARM C++ Linker
arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -O0 -fmessage-length=0 -ffunction-sections -fdata-sections -flto  -g3 -T "..\config\stm32f4ems\STM32F407xG_CCM.ld" -Xlinker --gc-sections -Wl,-Map,"rusefi.map" -Wl,-Map,"rusefi.map" -mcpu=cortex-m4 -mthumb -g3 -o "rusefi.elf"  ./util/LocalVersionHolder.o ./util/cli_registry.o ./util/crc.o ./util/cyclic_buffer.o ./util/data_buffer.o ./util/datalogging.o ./util/efilib.o ./util/efilib2.o ./util/histogram.o ./util/listener_array.o ./util/loggingcentral.o ./util/rfiutil.o  ./hw_layer/stm32f4/mpu_util.o ./hw_layer/stm32f4/stm32f4xx_hal_flash.o ./hw_layer/stm32f4/stm32f4xx_hal_flash_ex.o  ./hw_layer/serial_over_usb/usbcfg.o ./hw_layer/serial_over_usb/usbconsole.o  ./hw_layer/sensors/CJ125.o  ./hw_layer/mass_storage/usb_msd.o ./hw_layer/mass_storage/usb_msd_cfg.o  ./hw_layer/lcd/lcd_HD44780.o  ./hw_layer/HIP9011.o ./hw_layer/accelerometer.o ./hw_layer/adc_inputs.o ./hw_layer/board_test.o ./hw_layer/can_hw.o ./hw_layer/digital_input_hw.o ./hw_layer/flash.o ./hw_layer/gpio_helper.o ./hw_layer/hardware.o ./hw_layer/io_pins.o ./hw_layer/joystick.o ./hw_layer/max31855.o ./hw_layer/mcp3208.o ./hw_layer/microsecond_timer.o ./hw_layer/mmc_card.o ./hw_layer/neo6m.o ./hw_layer/pin_repository.o ./hw_layer/pwm_generator.o ./hw_layer/rtc_helper.o ./hw_layer/servo.o ./hw_layer/stepper.o ./hw_layer/trigger_input.o ./hw_layer/vehicle_speed.o  ./development/test/test.o ./development/test/testbmk.o  ./development/hw_layer/poten.o  ./development/engine_emulator.o ./development/engine_sniffer.o ./development/rfi_perftest.o ./development/sensor_chart.o ./development/trigger_emulator.o ./development/wave_analyzer.o  ./controllers/trigger/decoders/trigger_bmw.o ./controllers/trigger/decoders/trigger_chrysler.o ./controllers/trigger/decoders/trigger_gm.o ./controllers/trigger/decoders/trigger_honda.o ./controllers/trigger/decoders/trigger_mazda.o ./controllers/trigger/decoders/trigger_mitsubishi.o ./controllers/trigger/decoders/trigger_nissan.o ./controllers/trigger/decoders/trigger_rover.o ./controllers/trigger/decoders/trigger_subaru.o ./controllers/trigger/decoders/trigger_toyota.o ./controllers/trigger/decoders/trigger_universal.o  ./controllers/trigger/main_trigger_callback.o ./controllers/trigger/rpm_calculator.o ./controllers/trigger/spark_logic.o ./controllers/trigger/trigger_central.o ./controllers/trigger/trigger_decoder.o ./controllers/trigger/trigger_emulator_algo.o ./controllers/trigger/trigger_simulator.o ./controllers/trigger/trigger_structure.o  ./controllers/system/SingleTimerExecutor.o ./controllers/system/efiGpio.o ./controllers/system/event_queue.o ./controllers/system/pwm_generator_logic.o ./controllers/system/signal_executor_sleep.o  ./controllers/sensors/allsensors.o ./controllers/sensors/ego.o ./controllers/sensors/hip9011_lookup.o ./controllers/sensors/maf.o ./controllers/sensors/map.o ./controllers/sensors/thermistors.o ./controllers/sensors/tps.o ./controllers/sensors/voltage.o  ./controllers/math/biquad.o ./controllers/math/engine_math.o ./controllers/math/pid.o ./controllers/math/speed_density.o  ./controllers/core/EfiWave.o ./controllers/core/avg_values.o ./controllers/core/fsio_core.o ./controllers/core/fsio_impl.o ./controllers/core/interpolation.o ./controllers/core/signal_filtering.o ./controllers/core/table_helper.o  ./controllers/algo/accel_enrichment.o ./controllers/algo/advance_map.o ./controllers/algo/algo.o ./controllers/algo/auto_generated_enums.o ./controllers/algo/aux_pid.o ./controllers/algo/engine.o ./controllers/algo/engine_configuration.o ./controllers/algo/event_registry.o ./controllers/algo/fuel_math.o ./controllers/algo/idle_controller.o ./controllers/algo/lcd_menu_tree.o ./controllers/algo/malfunction_central.o ./controllers/algo/map_adjuster.o ./controllers/algo/nmea.o ./controllers/algo/signal_executor.o  ./controllers/PwmTester.o ./controllers/alternatorController.o ./controllers/electronic_throttle.o ./controllers/engine_controller.o ./controllers/error_handling.o ./controllers/flash_main.o ./controllers/idle_thread.o ./controllers/injector_central.o ./controllers/lcd_controller.o ./controllers/malfunction_indicator.o ./controllers/map_averaging.o ./controllers/obd2.o ./controllers/obd2viaCAN.o ./controllers/settings.o ./controllers/tachometer.o  ./console/fl_binary/fl_protocol.o  ./console/binary/tunerstudio.o ./console/binary/tunerstudio_io.o  ./console/console_io.o ./console/eficonsole.o ./console/status_loop.o  ./config/engines/GY6_139QMB.o ./config/engines/MiniCooperR50.o ./config/engines/acura_rsx.o ./config/engines/audi_aan.o ./config/engines/bmw_e34.o ./config/engines/chevrolet_c20_1973.o ./config/engines/chevrolet_camaro_4.o ./config/engines/citroenBerlingoTU3JP.o ./config/engines/custom_engine.o ./config/engines/daihatsu.o ./config/engines/dodge_neon.o ./config/engines/dodge_ram.o ./config/engines/dodge_stratus.o ./config/engines/engine_template.o ./config/engines/ford_1995_inline_6.o ./config/engines/ford_aspire.o ./config/engines/ford_festiva.o ./config/engines/ford_fiesta.o ./config/engines/geo_storm.o ./config/engines/gm_2_2.o ./config/engines/honda_600.o ./config/engines/honda_accord.o ./config/engines/lada_kalina.o ./config/engines/mazda_323.o ./config/engines/mazda_626.o ./config/engines/mazda_miata.o ./config/engines/mazda_miata_1_6.o ./config/engines/mazda_miata_nb.o ./config/engines/mazda_miata_vvt.o ./config/engines/mits [... cut ...]
lto1.exe: internal compiler error: in splice_child_die, at dwarf2out.c:4905
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
lto-wrapper.exe: fatal error: arm-none-eabi-g++ returned 1 exit status
compilation terminated.
c:/progra~1/gnu_tools/5.4_2016q3/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status
makefile:100: recipe for target 'rusefi.elf' failed
cs-make: *** [rusefi.elf] Error 1

rusefi, 2017-03-20:
hard
2017-03-20 18_34: EngineState: *** Compiled:     Mar 20 2017 - 18:34:01
2017-03-20 18_34: EngineState: COMPILER=5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
2017-03-20 18_34: EngineState: CH_FREQUENCY=1000
2017-03-20 18_34: EngineState: SERIAL_SPEED=115200
...

2017-03-20 18_34: EngineState: secondary trigger simulator: PD2 OM_DEFAULT phase=3
2017-03-20 18_34: EngineState: trigger error extra LED: NONE OM_DEFAULT
2017-03-20 18_34: EngineState: primary logic input: NONE
2017-03-20 18_34: EngineState: secondary logic input: NONE
2017-03-20 18_34: EngineState: zeroTestTime=4 maxHowFarOff=17814
2017-03-20 18_34: EngineState: advanceLookupTime=1090 now=-334982438 fuelCalcTime=1923
2017-03-20 18_34: EngineState: ignitionSchTime=1037 injectonSchTime=0
2017-03-20 18_34: EngineState: mapTime=4830/hipTime=1148/rpmTime=486/mainTriggerCallbackTime=5921
2017-03-20 18_34: EngineState: maxLockTime=0 / maxTriggerReentraint=0
2017-03-20 18_34: EngineState: maxEventQueueTime=144521
2017-03-20 18_34: EngineState: hipLastExecutionCount=1
2017-03-20 18_34: EngineState: hwSetTimerTime 294
2017-03-20 18_34: EngineState: totalTriggerHandlerMaxTime=143163

rusefi, 2017-03-21:
depends on https://github.com/rusefi/rusefi/issues/376

rusefi, 2017-03-21:
I do not see a difference in performance with build-in performance metrics:

2016q3 compiler

softfp
2017-03-20 23_21: EngineState: zeroTestTime=13 maxHowFarOff=32722
2017-03-20 23_21: EngineState: advanceLookupTime=2440 now=-1365003159 fuelCalcTime=6083
2017-03-20 23_21: EngineState: ignitionSchTime=2563 injectonSchTime=0
2017-03-20 23_21: EngineState: mapTime=8850/hipTime=1757/rpmTime=815/mainTriggerCallbackTime=12942
2017-03-20 23_21: EngineState: maxLockTime=0 / maxTriggerReentraint=0
2017-03-20 23_21: EngineState: maxEventQueueTime=335369
2017-03-20 23_21: EngineState: hwSetTimerTime 449
2017-03-20 23_21: EngineState: totalTriggerHandlerMaxTime=31305

2016q3 compiler

hard
2017-03-20 23_28: EngineState: zeroTestTime=13 maxHowFarOff=33544
2017-03-20 23_28: EngineState: advanceLookupTime=2492 now=1232060281 fuelCalcTime=6339
2017-03-20 23_28: EngineState: ignitionSchTime=2569 injectonSchTime=0
2017-03-20 23_28: EngineState: mapTime=8862/hipTime=1765/rpmTime=805/mainTriggerCallbackTime=13302
2017-03-20 23_28: EngineState: maxLockTime=0 / maxTriggerReentraint=0
2017-03-20 23_28: EngineState: maxEventQueueTime=338394
2017-03-20 23_28: EngineState: hwSetTimerTime 449
2017-03-20 23_28: EngineState: totalTriggerHandlerMaxTime=31054

rusefi, 2017-04-02:
Could it be about https://sourceforge.net/p/rusefi/tickets/149/ ? Because the natural logarithm issue is kind of back - intermittent issue with both current (ChibiOS2) and future (ChibiOS3) Makefile

fpoussin, 2017-04-02:
We should only use makefiles from now on.
Better for consistency :)

Also, for all math functions we should use the dsp library, as it will be included in the project, again better for consistency.

rusefi, 2017-04-05:
Mini-test https://github.com/rusefi/ChibiOS/tree/stable_log_issue/demos/STM32/RT-STM32F407-DISCOVERY-MEMS actually does not reproduce an issue.

Note that mini test does not have CPP  and C++ libraries, need to try that as next step.

fpoussin, 2018-12-25:
Is this still a thing?

rusefi, 2018-12-25:
@fpoussin yes i think so

fpoussin, 2018-12-25:
You should re-try with the current GCC version.
Also, your bug report on launchpad is still open: https://bugs.launchpad.net/gcc-arm-embedded/+bug/1674525

rusefi, 2018-12-25:
https://bugs.launchpad.net/gcc-arm-embedded/+bug/1674525 looks good - just posted there

rusefi, 2018-12-27:
@fpoussin just changing "softfp" to "hard" in the makefile causes linked error about "exit" missing

I hope it's not a huge thing but I can use some help here, can you please have a look?

fpoussin, 2018-12-27:
Which compiler are you using? is LTO enabled?

Current master is broken for me, can't test. (using GCC 8.2.1)

rusefi, 2018-12-27:
[image]
7.3.1

I see that Version 8-2018-q4-major was released December 20, 2018 a week ago?! can you fix master so that it starts to compile under this compiler? master status is green see https://github.com/rusefi/rusefi/commits/master

fpoussin, 2018-12-27:
There you go, PR open.

rusefi, 2018-12-27:
@fpoussin with GCC 8 fix hard now compiles for me with gcc 7 :) will try "hard" on a real vehicle in a bit

fpoussin, 2018-12-27:
Try LTO after that, it's been improved in 8.2

rusefi, 2018-12-29:
We now use "hard"! -glto should probably be a separate ticket, looks like not used at the moment,

closed 2018-12-29 by rusefi

rusefillc, 2020-10-04:
Are we using soft or hard now?

[image]

rusefillc, 2020-10-04:
[image]

commit a089d701 (2017-03-21): fixed #375
commit 8be71393 (2017-03-21): fixed #375
commit 3b6de354 (2018-12-29): fix #375