rusefi/rusefi #376: convert to less ancient gcc compiler

rusefi, 2017-03-20:
So far I was using discontinued Mentor/CodeSourcery G++ Lite from 2014. It's probably time to move to a more recent toolchain from https://launchpad.net/gcc-arm-embedded

See also https://github.com/rusefi/rusefi/issues/375

rusefi, 2017-03-20:
Will be using Latest version is 5-2016-q3-update

fpoussin, 2017-03-20:
Latest GCC here: (previously on launchpad)
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads

rusefi, 2017-03-21:
Building target: rusefi.elf
Invoking: Cross ARM C++ Linker
arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -O0 -fmessage-length=0 -ffunction-sections -fdata-sections  -g3 -T "..\config\stm32f4ems\STM32F407xG_CCM.ld" -Xlinker --gc-sections -Wl,-Map,"rusefi.map" -Wl,-Map,"rusefi.map" -mcpu=cortex-m4 -mthumb -g3 -o "rusefi.elf"  ./util/LocalVersionHolder.o ./util/cli_registry.o ./util/crc.o ./util/cyclic_buffer.o ./util/data_buffer.o ./util/datalogging.o ./util/efilib.o ./util/efilib2.o ./util/histogram.o ./util/listener_array.o ./util/loggingcentral.o ./util/rfiutil.o  ./hw_layer/stm32f4/mpu_util.o ./hw_layer/stm32f4/stm32f4xx_hal_flash.o ./hw_layer/stm32f4/stm32f4xx_hal_flash_ex.o  ./hw_layer/serial_over_usb/usbcfg.o ./hw_layer/serial_over_usb/usbconsole.o  ./hw_layer/sensors/CJ125.o  ./hw_layer/mass_storage/usb_msd.o ./hw_layer/mass_storage/usb_msd_cfg.o  ./hw_layer/lcd/lcd_HD44780.o  ./hw_layer/HIP9011.o ./hw_layer/accelerometer.o ./hw_layer/adc_inputs.o ./hw_layer/board_test.o ./hw_layer/can_hw.o ./hw_layer/digital_input_hw.o ./hw_layer/flash.o ./hw_layer/gpio_helper.o ./hw_layer/hardware.o ./hw_layer/io_pins.o ./hw_layer/joystick.o ./hw_layer/max31855.o ./hw_layer/mcp3208.o ./hw_layer/microsecond_timer.o ./hw_layer/mmc_card.o ./hw_layer/neo6m.o ./hw_layer/pin_repository.o ./hw_layer/pwm_generator.o ./hw_layer/rtc_helper.o ./hw_layer/servo.o ./hw_layer/stepper.o ./hw_layer/trigger_input.o ./hw_layer/vehicle_speed.o  ./development/test/test.o ./development/test/testbmk.o  ./development/hw_layer/poten.o  ./development/engine_emulator.o ./development/engine_sniffer.o ./development/rfi_perftest.o ./development/sensor_chart.o ./development/trigger_emulator.o ./development/wave_analyzer.o  ./controllers/trigger/decoders/trigger_bmw.o ./controllers/trigger/decoders/trigger_chrysler.o ./controllers/trigger/decoders/trigger_gm.o ./controllers/trigger/decoders/trigger_honda.o ./controllers/trigger/decoders/trigger_mazda.o ./controllers/trigger/decoders/trigger_mitsubishi.o ./controllers/trigger/decoders/trigger_nissan.o ./controllers/trigger/decoders/trigger_rover.o ./controllers/trigger/decoders/trigger_subaru.o ./controllers/trigger/decoders/trigger_toyota.o ./controllers/trigger/decoders/trigger_universal.o  ./controllers/trigger/main_trigger_callback.o ./controllers/trigger/rpm_calculator.o ./controllers/trigger/spark_logic.o ./controllers/trigger/trigger_central.o ./controllers/trigger/trigger_decoder.o ./controllers/trigger/trigger_emulator_algo.o ./controllers/trigger/trigger_simulator.o ./controllers/trigger/trigger_structure.o  ./controllers/system/SingleTimerExecutor.o ./controllers/system/efiGpio.o ./controllers/system/event_queue.o ./controllers/system/pwm_generator_logic.o ./controllers/system/signal_executor_sleep.o  ./controllers/sensors/allsensors.o ./controllers/sensors/ego.o ./controllers/sensors/hip9011_lookup.o ./controllers/sensors/maf.o ./controllers/sensors/map.o ./controllers/sensors/thermistors.o ./controllers/sensors/tps.o ./controllers/sensors/voltage.o  ./controllers/math/biquad.o ./controllers/math/engine_math.o ./controllers/math/pid.o ./controllers/math/speed_density.o  ./controllers/core/EfiWave.o ./controllers/core/avg_values.o ./controllers/core/fsio_core.o ./controllers/core/fsio_impl.o ./controllers/core/interpolation.o ./controllers/core/signal_filtering.o ./controllers/core/table_helper.o  ./controllers/algo/accel_enrichment.o ./controllers/algo/advance_map.o ./controllers/algo/algo.o ./controllers/algo/auto_generated_enums.o ./controllers/algo/aux_pid.o ./controllers/algo/engine.o ./controllers/algo/engine_configuration.o ./controllers/algo/event_registry.o ./controllers/algo/fuel_math.o ./controllers/algo/idle_controller.o ./controllers/algo/lcd_menu_tree.o ./controllers/algo/malfunction_central.o ./controllers/algo/map_adjuster.o ./controllers/algo/nmea.o ./controllers/algo/signal_executor.o  ./controllers/PwmTester.o ./controllers/alternatorController.o ./controllers/electronic_throttle.o ./controllers/engine_controller.o ./controllers/error_handling.o ./controllers/flash_main.o ./controllers/idle_thread.o ./controllers/injector_central.o ./controllers/lcd_controller.o ./controllers/malfunction_indicator.o ./controllers/map_averaging.o ./controllers/obd2.o ./controllers/obd2viaCAN.o ./controllers/settings.o ./controllers/tachometer.o  ./console/fl_binary/fl_protocol.o  ./console/binary/tunerstudio.o ./console/binary/tunerstudio_io.o  ./console/console_io.o ./console/eficonsole.o ./console/status_loop.o  ./config/engines/GY6_139QMB.o ./config/engines/MiniCooperR50.o ./config/engines/acura_rsx.o ./config/engines/audi_aan.o ./config/engines/bmw_e34.o ./config/engines/chevrolet_c20_1973.o ./config/engines/chevrolet_camaro_4.o ./config/engines/citroenBerlingoTU3JP.o ./config/engines/custom_engine.o ./config/engines/daihatsu.o ./config/engines/dodge_neon.o ./config/engines/dodge_ram.o ./config/engines/dodge_stratus.o ./config/engines/engine_template.o ./config/engines/ford_1995_inline_6.o ./config/engines/ford_aspire.o ./config/engines/ford_festiva.o ./config/engines/ford_fiesta.o ./config/engines/geo_storm.o ./config/engines/gm_2_2.o ./config/engines/honda_600.o ./config/engines/honda_accord.o ./config/engines/lada_kalina.o ./config/engines/mazda_323.o ./config/engines/mazda_626.o ./config/engines/mazda_miata.o ./config/engines/mazda_miata_1_6.o ./config/engines/mazda_miata_nb.o ./config/engines/mazda_miata_vvt.o ./config/engines/mitsubis [... cut ...]
lto1.exe: internal compiler error: in splice_child_die, at dwarf2out.c:5030
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
lto-wrapper.exe: fatal error: arm-none-eabi-g++ returned 1 exit status
compilation terminated.
c:/progra~1/gnu_tools/6_2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld.exe: error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status
makefile:101: recipe for target 'rusefi.elf' failed
cs-make: *** [rusefi.elf] Error 1

rusefi, 2017-03-21:
fails differently (manual merge of makefile and eclipse commands)

arm-none-eabi-gcc  ./util/LocalVersionHolder.o ./util/cli_registry.o ./util/crc.o ./util/cyclic_buffer.o ./util/data_buffer.o ./util/datalogging.o ./util/efilib.o ./util/efilib2.o ./util/histogram.o ./util/listener_array.o ./util/loggingcentral.o ./util/rfiutil.o  ./hw_layer/stm32f4/mpu_util.o ./hw_layer/stm32f4/stm32f4xx_hal_flash.o ./hw_layer/stm32f4/stm32f4xx_hal_flash_ex.o  ./hw_layer/serial_over_usb/usbcfg.o ./hw_layer/serial_over_usb/usbconsole.o  ./hw_layer/sensors/CJ125.o  ./hw_layer/mass_storage/usb_msd.o ./hw_layer/mass_storage/usb_msd_cfg.o  ./hw_layer/lcd/lcd_HD44780.o  ./hw_layer/HIP9011.o ./hw_layer/accelerometer.o ./hw_layer/adc_inputs.o ./hw_layer/board_test.o ./hw_layer/can_hw.o ./hw_layer/digital_input_hw.o ./hw_layer/flash.o ./hw_layer/gpio_helper.o ./hw_layer/hardware.o ./hw_layer/io_pins.o ./hw_layer/joystick.o ./hw_layer/max31855.o ./hw_layer/mcp3208.o ./hw_layer/microsecond_timer.o ./hw_layer/mmc_card.o ./hw_layer/neo6m.o ./hw_layer/pin_repository.o ./hw_layer/pwm_generator.o ./hw_layer/rtc_helper.o ./hw_layer/servo.o ./hw_layer/stepper.o ./hw_layer/trigger_input.o ./hw_layer/vehicle_speed.o  ./development/test/test.o ./development/test/testbmk.o  ./development/hw_layer/poten.o  ./development/engine_emulator.o ./development/engine_sniffer.o ./development/rfi_perftest.o ./development/sensor_chart.o ./development/trigger_emulator.o ./development/wave_analyzer.o  ./controllers/trigger/decoders/trigger_bmw.o ./controllers/trigger/decoders/trigger_chrysler.o ./controllers/trigger/decoders/trigger_gm.o ./controllers/trigger/decoders/trigger_honda.o ./controllers/trigger/decoders/trigger_mazda.o ./controllers/trigger/decoders/trigger_mitsubishi.o ./controllers/trigger/decoders/trigger_nissan.o ./controllers/trigger/decoders/trigger_rover.o ./controllers/trigger/decoders/trigger_subaru.o ./controllers/trigger/decoders/trigger_toyota.o ./controllers/trigger/decoders/trigger_universal.o  ./controllers/trigger/main_trigger_callback.o ./controllers/trigger/rpm_calculator.o ./controllers/trigger/spark_logic.o ./controllers/trigger/trigger_central.o ./controllers/trigger/trigger_decoder.o ./controllers/trigger/trigger_emulator_algo.o ./controllers/trigger/trigger_simulator.o ./controllers/trigger/trigger_structure.o  ./controllers/system/SingleTimerExecutor.o ./controllers/system/efiGpio.o ./controllers/system/event_queue.o ./controllers/system/pwm_generator_logic.o ./controllers/system/signal_executor_sleep.o  ./controllers/sensors/allsensors.o ./controllers/sensors/ego.o ./controllers/sensors/hip9011_lookup.o ./controllers/sensors/maf.o ./controllers/sensors/map.o ./controllers/sensors/thermistors.o ./controllers/sensors/tps.o ./controllers/sensors/voltage.o  ./controllers/math/biquad.o ./controllers/math/engine_math.o ./controllers/math/pid.o ./controllers/math/speed_density.o  ./controllers/core/EfiWave.o ./controllers/core/avg_values.o ./controllers/core/fsio_core.o ./controllers/core/fsio_impl.o ./controllers/core/interpolation.o ./controllers/core/signal_filtering.o ./controllers/core/table_helper.o  ./controllers/algo/accel_enrichment.o ./controllers/algo/advance_map.o ./controllers/algo/algo.o ./controllers/algo/auto_generated_enums.o ./controllers/algo/aux_pid.o ./controllers/algo/engine.o ./controllers/algo/engine_configuration.o ./controllers/algo/event_registry.o ./controllers/algo/fuel_math.o ./controllers/algo/idle_controller.o ./controllers/algo/lcd_menu_tree.o ./controllers/algo/malfunction_central.o ./controllers/algo/map_adjuster.o ./controllers/algo/nmea.o ./controllers/algo/signal_executor.o  ./controllers/PwmTester.o ./controllers/alternatorController.o ./controllers/electronic_throttle.o ./controllers/engine_controller.o ./controllers/error_handling.o ./controllers/flash_main.o ./controllers/idle_thread.o ./controllers/injector_central.o ./controllers/lcd_controller.o ./controllers/malfunction_indicator.o ./controllers/map_averaging.o ./controllers/obd2.o ./controllers/obd2viaCAN.o ./controllers/settings.o ./controllers/tachometer.o  ./console/fl_binary/fl_protocol.o  ./console/binary/tunerstudio.o ./console/binary/tunerstudio_io.o  ./console/console_io.o ./console/eficonsole.o ./console/status_loop.o  ./config/engines/GY6_139QMB.o ./config/engines/MiniCooperR50.o ./config/engines/acura_rsx.o ./config/engines/audi_aan.o ./config/engines/bmw_e34.o ./config/engines/chevrolet_c20_1973.o ./config/engines/chevrolet_camaro_4.o ./config/engines/citroenBerlingoTU3JP.o ./config/engines/custom_engine.o ./config/engines/daihatsu.o ./config/engines/dodge_neon.o ./config/engines/dodge_ram.o ./config/engines/dodge_stratus.o ./config/engines/engine_template.o ./config/engines/ford_1995_inline_6.o ./config/engines/ford_aspire.o ./config/engines/ford_festiva.o ./config/engines/ford_fiesta.o ./config/engines/geo_storm.o ./config/engines/gm_2_2.o ./config/engines/honda_600.o ./config/engines/honda_accord.o ./config/engines/lada_kalina.o ./config/engines/mazda_323.o ./config/engines/mazda_626.o ./config/engines/mazda_miata.o ./config/engines/mazda_miata_1_6.o ./config/engines/mazda_miata_nb.o ./config/engines/mazda_miata_vvt.o ./config/engines/mitsubishi.o ./config/engines/nissan_primera.o ./config/engines/rover_v8.o ./config/engines/sachs.o ./config/engines/saturn_ion.o ./config/engines/snow_blower.o ./config/engines/subaru.o ./config/engines/suzuki_vitara.o ./config/engines/test_engine.o ./config/engines/toyota_jzs147 [... cut ...] e -Wno-error=sign-compare -Wno-error=unused-parameter -Wno-error=missing-field-initializers -Werror=type-limits -Wno-error=strict-aliasing -Wno-error=attributes -ffunction-sections -fdata-sections -fno-common -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -fsingle-precision-constant -nostartfiles  -Wl,-Map=build/rusefi.map,--cref,--no-warn-mismatch,--library-path=chibios/os/ports/GCC/ARMCMx,--script=../config/stm32f4ems/STM32F407xG_CCM.ld,--gc-sections -mno-thumb-interwork -mthumb  -lm -o build/rusefi.elf

rusefi, 2017-03-21:
-g3 in linked command is the part which crashes linker - it works without it but crashes with it

also about 2K extra RAM usage with current GCC compiler, interesting

rusefi, 2017-03-21:
@fpoussin what is the politics/drama/etc between a bit older version at launchpad and newer version at developers.arm.com? I am trying to find the right place to report a bug.

Thank you for pointing at the newer gcc for arm!

rusefi, 2017-03-21:
Makefile set of parameters seems to work even with -g

But elicpse arm plugin setting are crashing linker if used with -g

https://bugs.launchpad.net/gcc-arm-embedded/+bug/1674525

closed 2017-03-21 by rusefi

fpoussin, 2017-03-22:
@rusefi For some reason they decided to stop publishing binaries on launchpad and use their own website instead, no idea why.

rusefi, 2017-03-22:
See also https://bugs.launchpad.net/gcc-arm-embedded/+bug/1674525

commit e94c43bd (2017-03-21): fix #376