rusefi/rusefi #380: better solution for board files

rusefi, 2017-03-22:
ST_STM32F4_DISCOVERY board files were modified under https://sourceforge.net/p/rusefi/tickets/293/

We should probably move our custom board files out of chibios folder

rusefi, 2017-03-22:
[image]

@fpoussin FIY :(

rusefi, 2017-03-22:
http://stackoverflow.com/questions/1405030/using-git-how-can-i-selectively-pull-merge-changes-from-anothers-fork

git remote add fpoussin https://github.com/fpoussin/rusefi.git
git fetch fpoussin
git log fpoussin/chibios-update
git cherry-pick -n 51e9e9dd9f67004e0a5c1ad6ae25ac6106473ef6

closed 2017-03-22 by rusefi

fpoussin, 2017-03-22:
Looks good :) 
I know, git can be rough sometimes...

commit cd877e9e (2017-03-22): fixed #380