rusefi/rusefi #382: datalog entries naming mess

rusefi, 2017-03-26:
Currently rusEfi produces three types of logs

Each type of logs has it's own name for same field, that's a mess and that would be improved

See https://github.com/rusefi/rusefi/tree/master/firmware/tunerstudio
See https://github.com/rusefi/rusefi/blob/master/java_console/models/src/com/rusefi/core/Sensor.java
See https://github.com/rusefi/rusefi/blob/master/firmware/console/status_loop.cpp

fpoussin, 2017-04-07:
I suggest we use Python + Jinja2 (template engine) + PyYaml for these.

rusefi, 2017-04-08:
Do not know anything about these tools unfortunatelly :(

I assume some code would be needed to calculate offsets - i doubt the tools would be flexible enough to generate TunerStudio .ini files structure, would they? Assuming some custom code would still be needed it's not too far from having everything in our own tools.

Anyway at this point all the tools are ready and the remaining part of the ticket is just to extract the constants and reference those on all three places - no infrastructure work left, only the actual content management.

rusefi, 2019-10-30:
things are good enough now

closed 2019-10-30 by rusefi