rusefi/rusefi #396: CPU usage stats with IRQ usage

rusefi, 2017-04-17:
ChibiOS has some means of tracking thread CPU usage but not much for IRQ ticks (?)

A lot of rusEfi logic is witin IRQ, we need some monitoring

rusefi, 2017-04-17:
@fpoussin  points at https://github.com/fpoussin/MotoLink/blob/master/code/app/main.c#L407

fpoussin, 2019-01-05:
You can probably close this one

rusefi, 2019-01-05:
Speaking of which, what is the ChibiOS way to access 168'000'000 value, i.e. CPU frequency?

// todo: access some existing configuration field
#define CORE_CLOCK 168000000

Yep, we now have this - see https://rusefi.com/forum/viewtopic.php?f=5&t=373&start=360#p30895

closed 2019-01-05 by rusefi

fpoussin, 2019-01-07:
I think they use HSI/HSE and PLL mult/div values from mcuconf.h

rusefi, 2019-01-07:
@fpoussin can you help with this?