rusefi/rusefi #397: reduce the pin output speed on the MCU as these are the major source of ripple on VDD.

rusefi, 2017-04-21:
@fpoussin says:
"If you reduce the GPIO speed (increase rise/fall times) you reduce the current each pin uses

so you sink less current from the ldo and you have less ripple

You can probably set most of them to 2Mhz

I think is only works for GPIO, not sure if it does for peripherals (AF mode)"

rusefi, 2017-04-21:
I can build a custom version with this change but the challenge is in validation :) Will ask Stefan

rusefi, 2017-04-21:
F0:
http://hertaville.com/stm32f0-gpio-tutorial-part-1.html

#define PIN_OSPEED_VERYLOW(n)       (0U << ((n) * 2U))
#define PIN_OSPEED_LOW(n)           (1U << ((n) * 2U))
#define PIN_OSPEED_MEDIUM(n)        (2U << ((n) * 2U))
#define PIN_OSPEED_HIGH(n)          (3U << ((n) * 2U))

rusefi, 2017-04-21:
@fpoussin says

basically anything that's not high speed com like USB or CAN can go to low speed
CAN can use medium which is like 50Mhz
you can set everything to 50Mhz for a start :slightly_smiling_face:
it's still half the drive current

rusefillc, 2021-01-22:
@mck1117 @andreika-git @dron0gus keep or close?