rusefi/rusefi #398: Firmware Update via UART

andreika-git, 2017-04-21:
Add support for firmware uploading in rusEFI console, and firmware updating in rusEFI firmware code.

rusefi, 2017-04-21:
See also #147

andreika-git, 2017-04-21:
This might help...

fpoussin, 2017-04-23:
See https://github.com/fpoussin/MotoLink/tree/master/code/bootloader
It uses USB but it's exactly the same if using CDC.

Bootloader is updated via USB DFU (tactile button on boot0 pin)
This can also be a solution as it does not involve any firmware changes.

rusefi, 2019-09-15:
http://dfu-util.sourceforge.net/ to provide reset after DFU upload

fpoussin, 2019-09-15:
Looks like the libusb driver is not installed

rusefi, 2019-09-16:
no idea how to install it, is there a command i can use?

rusefi, 2019-09-16:
ps: it detects the HW ID so it must be communicating?

fpoussin, 2019-09-16:
Did you read the documentation?

rusefi, 2019-09-16:
Obviously not, anyone ever does that?!

fpoussin, 2019-09-16:
Use https://zadig.akeo.ie to install the WinUsb driver and try again.

rusefi, 2020-06-26:
@ksmola any interest?

rusefi, 2020-06-26:
On the other hand we have https://github.com/NordicSemiconductor/Android-DFU-Library

Which could be potentially converted into non-Android JAVA DFU uploader

FYI @andreika-git

rusefi, 2020-06-28:
See also https://github.com/rusefi/dfu_java

rusefillc, 2022-11-20:
Sounds like OpenBLT also "via CAN" is the fanciest option and we now have it.

closed 2022-11-20 by rusefillc