rusefi/rusefi #408: Help with Eclipse Rusefi C project

filipk5, 2017-05-03:
I am trying to get this project going in Eclipse environment. I can not get it to run due to an error 127. Thanks for the help.
[image]

rusefi, 2017-05-03:
http://rusefi.com/forum/viewtopic.php?f=5&t=9 would be the most relevant forum thread

First troubleshooting step from there is
arm-none-eabi-gcc -v
in command prompt, what does it show?

I believe your Eclipse does not see arm-none-eabi-gcc installed

closed 2017-05-03 by rusefi

filipk5, 2017-05-03:
Thanks 

I posted it it somewhere else in github but it looks like it does not need ti be there.

Help with Eclipse Rusefi C project · Issue #408 · rusefi/rusefi

  
|  
|   
|   
|   |    |

   |

  |
|  
|    |  
Help with Eclipse Rusefi C project · Issue #408 · rusefi/rusefi
 I am trying to get this project going in Eclipse environment. I can not get it to run due to an error 127. Thank...  |   |

  |

  |

 

Filip