rusefi/rusefi #421: convert serial port to DMA

rusefi, 2017-05-23:
In order to get reliable serial port we should migrate to DMA mode

rusefi, 2017-05-23:
See http://rusefi.com/forum/viewtopic.php?f=15&t=1233&start=30

See http://www.chibios.com/forum/viewtopic.php?f=25&t=3441&start=10

rusefi, 2017-05-23:
https://github.com/rusefi/ChibiOS/pull/1

rusefi, 2017-05-23:
https://github.com/rusefi/rusefi/pull/422/

rusefi, 2017-05-23:
[image]
@andreika-git ops

andreika-git, 2017-05-24:
A conflict of USART3 and SPI2. (STM32_SPI_SPI2_RX_DMA_STREAM == STM32_UART_USART3_TX_DMA_STREAM).

rusefi, 2017-05-24:
https://github.com/rusefi/rusefi/pull/425

andreika-git, 2017-05-24:
It seems that the conflict is generally unsolvable for STM32F4:
http://www.chibios.com/forum/viewtopic.php?t=336#p2843
http://www.chibios.com/forum/viewtopic.php?t=1240#p11334
It's a hardware limitation (see "DMA1 request mapping" in RM0090 reference manual).

The good news is it's USART3_TX alone that's causing the conflict, not Rx. And we mostly need DMA for receiving. So the only way I see in this situation - is to use DMA only for USART3-Rx, and 'Serial' interrupts for USART3-Tx. It's like a hybrid of Serial and Uart drivers - Rx half from Uart and Tx half from Serial... Maybe we need to create a separate UART driver with our new DMA code, which would have two switchable modes - full-dma and 'hybrid'. 
As for now, users will have to choose between SPI2 (the default) and USART3-DMA in their configs...

rusefi, 2017-05-24:
#define STM32_SPI2_RX_DMA_MSK               STM32_DMA_STREAM_ID_MSK(1, 3)
#define STM32_SPI2_TX_DMA_MSK               STM32_DMA_STREAM_ID_MSK(1, 4)

#define STM32_USART3_RX_DMA_MSK             STM32_DMA_STREAM_ID_MSK(1, 1)

#define STM32_USART3_TX_DMA_MSK             (STM32_DMA_STREAM_ID_MSK(1, 3) |\
                                             STM32_DMA_STREAM_ID_MSK(1, 4))

mck1117, 2021-02-17:
this is done right? we use the uart driver.

closed 2021-02-17 by rusefillc