rusefi/rusefi #441: performance: general performance work

rusefi, 2017-06-08:
time for a fresh look at performance
- what are the sources of lock contention, if any?
- what are the bottlenecks?
- are there any places where we waste CPU?

rusefi, 2017-06-09:
See also #375

rusefi, 2017-06-09:
Just for the record: release firmware

2017-06-08 23_55: EngineState: advanceLookupTime=566 now=-728007371 fuelCalcTime=3913
2017-06-08 23_55: EngineState: ignitionSchTime=1275 injectonSchTime=0
2017-06-08 23_55: EngineState: mapTime=4030/hipTime=938/rpmTime=389/mainTriggerCallbackTime=3206
2017-06-08 23_55: EngineState: maxEventCallbackDuration=13042
2017-06-08 23_55: EngineState: hipLastExecutionCount=1
2017-06-08 23_55: EngineState: hwSetTimerDuration=253
2017-06-08 23_55: EngineState: totalTriggerHandlerMaxTime=11604
2017-06-08 23_55: EngineState: maxPrecisionCallbackDuration=14049

mck1117, 2021-11-29:
not actionable, performance is miles ahead of where it used to be

closed 2021-11-29 by mck1117