rusefi/rusefi #454: publish continues integration artifacts online

rusefi, 2017-06-29:
@andreika-git  would love to see full jenkins error report

@fpoussin what would you suggest? at the moment I have my jenkins instance with real stm32 hardware in my apparment and I would rather not expose my IP address for paranoia reasons, is there anything to publish jenkins state on the web? I can upload to the rusefi for example

fpoussin, 2018-12-25:
You can store releases on Amazon S3 free, there are tons of APIs to make this automatic.

https://wiki.jenkins.io/display/JENKINS/S3+Plugin

fpoussin, 2018-12-25:
Sure, PM me on slack.

fpoussin, 2018-12-26:
https://developer.github.com/v3/repos/releases/#upload-a-release-asset

rusefi, 2019-04-13:
This is still needed and I still do not have brain cycles to implement, I do not even understand the scope of work - is this a 3 minute project or a 30 minutes project?

@fpoussin @dron0gus @mck1117 @andreika-git as a simpler workaround I can simply add your emails into CI report destination list. Please respond if you are interested

rusefi, 2020-06-19:
Closing this since we seem to be moving towards GH Actions

closed 2020-06-19 by rusefi