rusefi/rusefi #455: RPM hard limit depending on CLT

rusefi, 2017-06-29:
Variable RPM hard limit depending on engine temperate would be a nice feature right from Lexus LFA

See https://www.youtube.com/watch?v=PopzpO-Iu94

AlexHLinS, 2021-01-24:
I've discovered the idea and current map's set. 
Can we use the Warmup fuel multiplier for these? The model will be: [RPM Limit] = [RPM hard limit] / [CLT Multipler] (e.g. engine has a red line at 7000 rpm, CLT multiplier at 0 Celsius equal 1.28 => rpm limit set to 7k / 1.28 = 5,5k rpm) 
I  offer to add to the "LimpManager" method named as setCLTRevLimit(bool CLRRpmTag). This one will check if current m_faultRevLimit, and set it to calculated only if calculated is less than existing.
If you applying my idea, I'll go be going to do this.

rusefillc, 2021-01-24:
i would not mix Warmup and hard limit, those are separate things to be configured separately

we need a new curve here, could be a small 4 element curve

AlexHLinS, 2021-01-24:
Ok, got it.

mck1117, 2022-06-22:
with #4267, this is ddddone!

closed 2022-06-22 by mck1117