rusefi/rusefi #467: Timing IAT Correction broken?

andreika-git, 2017-08-04:
[image]
[2017-08-02_09.09.13.msl.zip](https://github.com/rusefi/rusefi/files/1200223/2017-08-02_09.09.13.msl.zip)
(Starting from 719 sec)

typedef Map3D<IGN_RPM_COUNT, IGN_LOAD_COUNT, float> ign_Map3D_t;
static ign_Map3D_t iatAdvanceCorrectionMap("iat corr");

iatAdvanceCorrectionMap.init(config->ignitionIatCorrTable, config->ignitionIatCorrLoadBins,
   config->ignitionIatCorrRpmBins);
iatCorrection = iatAdvanceCorrectionMap.getValue((float) rpm, engine->sensors.iat);

mck1117, 2020-04-02:
I think this got fixed with #945

mck1117, 2021-01-03:
this is fixed, we can close

closed 2021-01-03 by rusefillc