rusefi/rusefi #469: configuration override in case of reading error

rusefi, 2017-08-05:
Historically rusEfi would override configuration with defaults if configuration was not read successfully.

It's a very open question if that's the smartest reaction to a failed config read. What if failed config read was caused by low voltage or some EMI spike?

It's very unclear what's going on, but it's probably better strategy would be not to override the configuration with the defaults since these are not really useful for any real user

rusefi, 2017-08-05:
See also #341

stefanst, 2017-08-05:
Agreed that overriding the whole configuration is not helpful. 
Is there sufficient memory to store two configs? If so, we could switch to backup-config.

mck1117, 2021-03-28:
Backup config storage didn't actually work, but was fixed long ago.  Closing.

closed 2021-03-28 by mck1117