rusefi/rusefi #487: URGENT: mono-injection support

rusefi, 2017-11-06:
At the moment rusEfi assumes that each cylinder has it's individual injector

we need to add support for single injector ASAP

rusefi, 2017-11-07:
IM_SINGLE_POINT

rusefi, 2017-11-07:
A problem is that historically some calculations were done on a per-cylinder basis. A proper change would be to convert everything to per-engine basis and only divide amounts by the number of cylinders if needed

mck1117, 2017-11-07:
A temporary hack that works is to set the displacement wrong so that the math is done for each cylinder having the displacement of the whole engine.  For example, a 2 liter with a single 800cc injector would be entered as an 8 liter with 800cc injectors, then only plug in a single injector channel.

closed 2017-11-11 by rusefi