rusefi/rusefi #492: auto-tune injector lag

rusefi, 2017-11-16:
See https://rusefi.com//forum/viewtopic.php?f=5&t=1085

rusefi, 2017-11-20:
set injection_mode X commands are still not causing engine to stall (AFR not nice since two_wire batch not enabled for this test)
[2017-11-20_00_05_30.zip](https://github.com/rusefi/rusefi/files/1486506/2017-11-20_00_05_30.zip)

andreika-git, 2017-11-20:
Another test of inj.mode switch in real-time. It works:
[2017-11-20_17.27.40_inj_mode_switch_test.msl.zip](https://github.com/rusefi/rusefi/files/1488644/2017-11-20_17.27.40_inj_mode_switch_test.msl.zip)

[image]

rusefi, 2018-01-18:
Fresh test logs - now with injector lag (under @@fuel_something broken column name) and "mode" for "Engine Mode"

https://svn.code.sf.net/p/rusefi/code/misc/logs/2003_mazda_miata/2018-01-17_21.35.46_issue_492_injector_lag.msl

rusefillc, 2025-08-12:
tagging @ggurov