rusefi/rusefi #501: PID: idle valve position after deccel

metra52, 2017-11-29:
Today if we turn on the PID idle valve, we have an idle valve position = PID Min after decceleration, but it's not right.
We must save the last idle position before acceleration or TPS position more PID TPS Deactivation and move idle valve to the saved last position.

rusefi, 2017-11-29:
this is about idlePidDeactivationTpsThreshold but looks like it's the current logic already?

	if (getTPS(PASS_ENGINE_PARAMETER_SIGNATURE) > boardConfiguration->idlePidDeactivationTpsThreshold) {
		// just leave IAC position as is
		return currentIdlePosition;
	}

rusefi, 2017-11-29:
@metra52  please attach a log showing a drop in idlePosition

stefanst, 2017-11-29:
Last idle valve position held is not always the correct one to fall back to. Engine conditions may have changed in the meantime. Idle Valve should not return to last known position, but to an "initial values" table. Initial values should be based on coolant temp and desired idle RPM. A modifier for manifold air temp would also be useful if the value is available.

metra52, 2017-11-29:
[image]

the first push on the gas (green color,) the regulator drops from 20 percent to 5 (red line) and the RPM fall down (yellow line) then I throttle pressed so that the machine does not stoped

metra52, 2017-11-29:
@stefanst in the rusefi firmware not table with initial values for idle valve. in this case we hav only manual position and CLT multiplier

metra52, 2017-11-29:
on image we saw that idle valve position after decceleration TPS change to more less than before acceleration

metra52, 2017-11-29:
View from 898sec. This is log file
[cold start.zip](https://github.com/rusefi/rusefi/files/1515785/cold.start.zip)

metra52, 2017-11-29:
CurrentTune
[CurrentTune.zip](https://github.com/rusefi/rusefi/files/1515796/CurrentTune.zip)

andreika-git, 2018-03-29:
#589

mck1117, 2021-01-03:
this was fixed long ago

closed 2021-01-03 by rusefillc