rusefi/rusefi #509: Document rusEfi changes to Chibios & attach a patch

rusefi, 2017-12-03:
In light of yet another ChibiOS migration we need to document all areas of changes rusEfi is making to ChibiOS and produce the current version of a patch file

rusefi_chibios_16.patch

rusefi, 2017-12-03:
See also #463

rusefi, 2017-12-03:
@andreika-git any chance you can contribute the rusefi_chibios_16.patch file?

andreika-git, 2017-12-03:
I'll try my best

andreika-git, 2017-12-04:
Here we are:
[rusefi_custom_chibios.txt](https://github.com/rusefi/rusefi/files/1527661/rusefi_custom_chibios.txt)

[rusefi_chibios_16.patch.zip](https://github.com/rusefi/rusefi/files/1527664/rusefi_chibios_16.patch.zip)

rusefi, 2018-01-23:
See also https://github.com/rusefi/rusefi/issues/65

rusefi, 2018-01-25:
We had a good grip in this in firmware/docs folder

closed 2018-01-25 by rusefi