rusefi/rusefi #517: does not compile with gcc-arm-none-eabi-7-2017-q4-major

rusefi, 2017-12-22:
rusefi firmware does not compile with latest gcc-arm-none-eabi-7-2017-q4-major

Compiling chschd.c
ChibiOS3/os/rt/src/chschd.c: In function 'wakeup':
ChibiOS3/os/rt/src/chschd.c:290:5: error: this statement may fall through [-Werror=implicit-fallthrough=]
     chSemFastSignalI(tp->p_u.wtsemp);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ChibiOS3/os/rt/src/chschd.c:294:3: note: here
   case CH_STATE_WTCOND:
   ^~~~
cc1.exe: all warnings being treated as errors
make: *** [ChibiOS3/os/common/ports/ARMCMx/compilers/GCC/rules.mk:218: build/obj/chschd.o] Error 1

rusefi, 2018-07-28:
time to try version 2018 and migrate maybe at least to 2017

closed 2018-07-31 by rusefi

mck1117, 2018-08-02:
This broke the build on my machine.  

`matthew@dippen:/mnt/c/Users/matth/GitHub/rusefi2/firmware$ make
Compiling crt0_v7m.S
cc1: error: -Werror=implicit-fallthrough: no option -Wimplicit-fallthrough
cc1: error: -Werror=bool-operation: no option -Wbool-operation
ChibiOS4/os/common/startup/ARMCMx/compilers/GCC/rules.mk:240: recipe for target 'build/obj/crt0_v7m.o' failed
make: *** [build/obj/crt0_v7m.o] Error 1
matthew@dippen:/mnt/c/Users/matth/GitHub/rusefi2/firmware$ arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/4.9.3/lto-wrapper
Target: arm-none-eabi
Configured with: ../src/configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/lib/include' --mandir='/usr/lib/share/man' --infodir='/usr/lib/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --mandir=/usr/share/man --enable-languages=c,c++,lto --enable-multilib --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --build=x86_64-linux-gnu --target=arm-none-eabi --with-system-zlib --with-gnu-as --with-gnu-ld --with-pkgversion=15:4.9.3+svn231177-1 --without-included-gettext --prefix=/usr/lib --infodir=/usr/share/doc/gcc-arm-none-eabi/info --htmldir=/usr/share/doc/gcc-arm-none-eabi/html --pdfdir=/usr/share/doc/gcc-arm-none-eabi/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --disable-libstdc++-v3 --host=x86_64-linux-gnu --with-headers=no --without-newlib --with-multilib-list=armv6-m,armv7-m,armv7e-m,armv7-r CFLAGS='-g -O2 -fstack-protector-strong' CPPFLAGS=-D_FORTIFY_SOURCE=2 CXXFLAGS='-g -O2 -fstack-protector-strong' FCFLAGS='-g -O2 -fstack-protector-strong' FFLAGS='-g -O2 -fstack-protector-strong' GCJFLAGS='-g -O2 -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro' OBJCFLAGS='-g -O2 -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fstack-protector-strong' INHIBIT_LIBC_CFLAGS=-DUSE_TM_CLONE_REGISTRY=0 AR_FOR_TARGET=arm-none-eabi-ar AS_FOR_TARGET=arm-none-eabi-as LD_FOR_TARGET=arm-none-eabi-ld NM_FOR_TARGET=arm-none-eabi-nm OBJDUMP_FOR_TARGET=arm-none-eabi-objdump RANLIB_FOR_TARGET=arm-none-eabi-ranlib READELF_FOR_TARGET=arm-none-eabi-readelf STRIP_FOR_TARGET=arm-none-eabi-strip
Thread model: single
gcc version 4.9.3 20150529 (prerelease) (15:4.9.3+svn231177-1)`

mck1117, 2018-08-02:
Also, I'm not sure we really want to turn off those two warnings.  Allowing case fallthrough is pretty dangerous.  Exhibit A, Apple's goto fail; goto fail; bug from a few years ago.

Likewise, turning off warnings for suspicious operations on boolean types isn't a great idea.

Both of the warnings this turns off are things we do genuinely want to be warned about.  If one of these is emitted, fix the code instead of turning off the warning.

rusefi, 2018-08-02:
@mck1117 you are on compiler version 4 from what year?

looks like current compiler is something like version 7?

As for warnings, please note that the errors are within ChibiOS code. I do not have resources to maintain my own fork of ChibioOS with these warnigs addressed nor do I have resources to convince the authors to make changes without me. 

I am open for suggestions how to improve things.

mck1117, 2018-08-02:
Looks like the build is from May 2015.  So it's reasonably old.  They must have changed something, because omitting the two options that my compiler doesn't understand still compiles fine.

Is it possible to set different compiler options for ChibiOS and for the rest of rusefi? We could set it a bit less strict on ChibiOS, but then turn it up for the rusefi stuff.

rusefi, 2018-08-02:
@mck1117 https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads

mck1117, 2018-08-02:
Windows or *nix?

rusefi, 2018-08-02:
@mck1117 Windows in my case but really should not matter?