rusefi/rusefi #533: F32 curve values from controller - unable to edit?

rusefi, 2018-01-10:
Attached project and tune. Note (invalid) NaN values from controller

<constant cols="1" digits="0" name="cltIdleRpm" rows="16">
         NaN 
         NaN 
         NaN 
         NaN 
         NaN 
         NaN 
         NaN 
         NaN 
         2500.0 
         2000.0 
         1500.0 
         1400.0 
         1100.0 
         900.0 
         900.0 
         900.0 
      </constant>

looks like there is no way to edit these via TS?

We can handle on our end by making sure that no curve has NaN values and override NaN with zeros, but maybe users can simply edit tune files and replace NaN with zero?

rusefi, 2018-01-10:
Emailed Phil

1) I have a curve with F32 bins.
2) my controller had NaN F32 values by mistake
3) I would read tune with NaN from controller
4) actual result: unable to edit NaN values. these NaN values are displayed as zero
4) expected result: NaN displayed as NAN, I can edit from NaN into whatever.

mck1117, 2021-02-13:
I don't think this exists any more, close?

closed 2021-02-13 by rusefillc