rusefi/rusefi #538: MAF2MAP convestion

rusefi, 2018-01-20:
On 1.6 miata we have a MAF sensor with a known translation table from MAF and RPM into MAP.

It should be possible to run a car using this table.

See http://miatajoe.50megs.com/index.html

rusefi, 2018-01-20:
[AFM Removal from the 1.6 liter Miata using eManage Ultimate.pdf](https://github.com/rusefi/rusefi/files/1649210/AFM.Removal.from.the.1.6.liter.Miata.using.eManage.Ultimate.pdf)
[afm_map.txt](https://github.com/rusefi/rusefi/files/1649212/afm_map.txt)

rusefi, 2018-01-21:
See also http://mdac.com.au/2017/12/10/air-flow-meters-everything-you-need-to-know-for-diy-efi/

mck1117, 2018-08-02:
What's wrong with doing the normal thing and running the car with speed-density? Yes, it requires a retune of the VE table, but that's not a big deal.  The MAP->MAF hack was just that, a hack so that you could run without a MAF sensor on a stock ECU.  We are not using a stock ECU.  The math in that table can be pretty easily transformed in to a VE map to run standard speed density, so why does this exist?

rusefi, 2018-08-24:
See also https://rusefi.com/forum/viewtopic.php?f=5&t=1413

rusefi, 2019-09-08:
[AFM Removal from the 1.6 liter Miata using eManage Ultimate.pdf](https://github.com/rusefi/rusefi/files/3587953/AFM.Removal.from.the.1.6.liter.Miata.using.eManage.Ultimate.pdf)

mck1117, 2020-05-28:
I think this can close now that we have NA Miata flappy door MAF transfer function

closed 2020-05-28 by rusefi