rusefi/rusefi #553: Virtual serial on MacOS

rusefi, 2018-01-25:
puff reports that something is not right with virtual serial on some version of MacOS

1) Puff, which exact version of MacOS?
2) What is the problem exactly - does it affect rusEfi console only, or TunerStudio only, or both?

thirstyone, 2018-01-25:
1. OSX 10.11.6, El Capitan. Java 1.8.0_161. 
2. The problem is with rusEfi console only, TunerStudio works just fine. 
The pattern is following: 
`Sending packet 47 
text header: waiting for 2 byte(s)
chartPanel size java.awt.Dimension[width=1415,height=770]
imagePanel getPreferredSizejava.awt.Dimension[width=1415,height=10]
chartPanel size java.awt.Dimension[width=1415,height=770]
imagePanel getPreferredSizejava.awt.Dimension[width=1415,height=10]
text body: waiting for 5435 byte(s)
Unexpected pending data: 5371 byte(s)`
Basically, after it says "Sending packet 47", it receives 64/96/128/256 bytes less than expected.

rusefi, 2018-01-25:
@thirstyone  can you please attach the complete log file?

thirstyone, 2018-01-25:
https://pastebin.com/9WMnvgMb

rusefi, 2018-01-25:
I've made some minor changes to logs in 20180125 but no idea on root cause. Here we have a huge start-up packet which TS does not use :(

rusefi, 2018-01-25:
@thirstyone any idea what driver is your OSX 10.11.6 using for this port?

thirstyone, 2018-01-25:
I have no idea of the driver version. Is there a chance to output it to console during console startup?

thirstyone, 2018-01-25:
Thu Jan 25 12:19:05 MSK 2018<EOT>: output channels header: waiting for 2 byte(s)
Thu Jan 25 12:19:05 MSK 2018<EOT>: output channels body: waiting for 132 byte(s)
Thu Jan 25 12:19:05 MSK 2018<EOT>: Unexpected pending data: 58 byte(s)

rusefi, 2018-01-25:
@thirstyone are you on a per-byte internet plan? :)

please let me see the WHOLE log, it might help me.

mck1117, 2021-02-13:
USB <--> serial works fine on mac these days, close issue?

closed 2021-02-13 by rusefillc