rusefi/rusefi #568: MAP briefly drops to 0 while cranking

darxfame, 2018-02-06:
MAP briefly drops to 0 while cranking. Мап по какой то непонятной причине каждый раз на запуске падает в положение 0. Ниже приведен скриншот и лог файл с двумя попытками запуска. Не стоит обращать внимание на успешность попыток, это никоим образом не влияет на проявление данной проблемы. Проблема в логе зафиксирована на метке 23.563 а так же 47.047
[image]
[2018-02-02_05.56.26.zip](https://github.com/rusefi/rusefi/files/1700622/2018-02-02_05.56.26.zip)

rusefi, 2018-02-06:
@darxfame нужен currenttune. например длинна усреднения какая?

к сожалению почти невозможно такое починить на расстоянии - как бы доказать, что это не электрическая проблема для начала?

сейчас гляну, какие у нас варианты записи исходного значения raw map

rusefi, 2018-02-06:
https://rusefi.com/wiki/index.php?title=Manual:Debug_fields
нужен лог в режиме ADC - там будет исходный сигнал MAP без обработки

darxfame, 2018-02-06:
@rusefi к сожалению такой же эффект проявлялся и у @andreika-git и тут явно нет электрической проблемы. Лог как найду предоставлю. усреднение не используется в настройке стоит 0. От смены окна ДАД эффект не пропадает, и присутствует он только в районе 200 оборотов

darxfame, 2018-02-06:
Прикладываю лог
[2018-01-30_14.09.55.zip](https://github.com/rusefi/rusefi/files/1700712/2018-01-30_14.09.55.zip)

rusefi, 2018-02-06:
я добавил debug mode 27 = DBG_MAP чтоб больше данных собрать

ну а читая map_averaging.cpp я вижу, что в первый раз когда появляются обороты currentPressure будет нулём, пока не будет первого замера. но странно, что ноль есть и второй запуск после остановки - этого я понять не могу. или там была перезагрузка ЭБУ?

@andreika-git

darxfame, 2018-02-06:
@rusefi да, была перезагрузка ЭБУ, но вот то что MAP падает в 0 это в корне неверно, у нас же на запуске используется таблица injection phase и когда MAP падает в 0 то это сбивает фазу впрыска

rusefi, 2018-02-07:
@darxfame ты 100% прав, что MAP падать до нуля не должен. итого - примерно в чём дефект понятно, починить будет достаточно просто!

darxfame, 2018-02-07:
@rusefi 
Проблема не исправлена 
[2018-02-07_11.00.28 map.zip](https://github.com/rusefi/rusefi/files/1702612/2018-02-07_11.00.28.map.zip)

andreika-git, 2018-02-07:
Есть догадка, что это из-за boardConfiguration->mapMinBufferLength=0. При этом mapMinBufferLength=1, из mapAveragingTriggerCallback() вызывается applyMapMinBufferLength() и стирает averagedMapRunningBuffer...

andreika-git, 2018-02-08:
`	if (!cisnan(minPressure))
		currentPressure = minPressure;
`
seems to work?..

rusefi, 2018-02-09:
@andreika-git но можешь ли ты объяснить почему? я не хотел бы затыкать проблему, не понимая её природы :(

andreika-git, 2018-02-09:
Пока лишь могу сказать, что на 1-м цикле mapMeasurementsCounter=0. Т.е. startAveraging() и endAveraging() вызываются, а вот mapAveragingAdcCallback() между ними - не вызывается.

andreika-git, 2018-02-10:
Вообще, сама идеология текущей реализации окон MAP подразумевает, что ширина окна задаётся пользователем в углах - и это не даёт никакой гарантии того, что будет хотя бы 1 измерение за такт...
Я бы предложил добавить альтернативный вариант задания ширины окна в количестве N замеров на такт (где Nmin=1).

rusefi, 2018-02-10:
сделал проверку, что не делим на ноль mapMeasurementsCounter

но не понимаю, как бы у нас там был ноль - не понимаю тебя, что значит первый цикл? я думал, там всегда будет сколько-то замеров и всегда будет всё хорошо

@andreika-git  @darxfame можно попросить проверить ещё раз?

PS: а в логах как раз v_averagedMapValue = debugFloatField1 = NAN

т.е. проблема где-то рядом и мы её затыкаем проверкой на ноль, вот только мы теряем замер из-за этого. в идеале понять, как у нас оказывается mapMeasurementsCounter = 0

rusefi, 2018-03-04:
@darxfame says thiis is resolved now.

closed 2018-03-04 by rusefi