rusefi/rusefi #587: PID fuel correction is limited to ±0ms.

mck1117, 2018-03-25:
fuelClosedLoopPid.maxValue and fuelClosedLoopPid.minValue are never set, so they're both zero.  The PID logic then clamps the value to 0, so no correction is ever made, even when the conditions in the "Fuel short-term closed loop correction" page are satisfied.

rusefi, 2018-03-25:
Does https://github.com/rusefi/rusefi/commit/654784406585ce0d168ccd9cfdf195d08190c01e help?

mck1117, 2018-03-25:
Yep! That looks just like 446ba977d0acdd196149b28e402daf38e00e1efd

mck1117, 2018-03-25:
Ok, part B: the allowed min/max value for the PID controller min/max in tunerstudio are 0 and 3000 respectively, but they should probably be -3000 to 3000.

See c8504a38748ae741b3002f3026b5205b34f68295.

rusefi, 2018-03-25:
See https://github.com/rusefi/rusefi/commit/ac5401fc4719068c0ea2df66aa26c9b5c6f10efa

mck1117, 2018-03-26:
Sweet! And to your comment, I changed the significant figures to 4 since at 7 the gains don't fit properly in the text boxes in TunerStudio.
[image]
Was there a case where that many significant figures were actually needed?

mck1117, 2018-03-26:
Anyway, this fixed the problem so I'm closing the issue.

closed 2018-03-26 by mck1117

rusefi, 2018-03-26:
@mck1117 yes I believe 7 digits were needed - this was more about scale less about precision. Do not remember the exact use-case but I have a feeling it's there for a reason. Hopefully one day I will document why.