rusefi/rusefi #593: PID parameter (names) swapped in the CJ125 heating code?

martonmiklos, 2018-05-26:
Hey,

Thanks for sharing this great project!
I am working on a CJ125 based interface bringup and came across your code. 

Here:
https://github.com/rusefi/rusefi/blob/master/firmware/hw_layer/sensors/CJ125.cpp#L455

you pass the vUr (current Nerst value) as a target (setpoint) and vUrCal as the input (current) value. However I think the PID controller's setpoint should be the vUrCal. 

By tracing down the code I got to the 
```
float Pid::getRawValue(float target, float input, float dTime)
```
https://github.com/rusefi/rusefi/blob/master/firmware/controllers/math/pid.cpp#L41

Could you please take a look on it, I am unsure if I did not missed something.

rusefi, 2018-05-27:
@andreika-git what do you think?

andreika-git, 2018-05-27:
PID doesn't care about the target or the input, it knows only the error value as the difference of (target - input). and if we swap them we'll just get a sign inversion. If target=vUrCal, and input=vUr, then error=vUrCal-vUr, i.e. if vUr<vUrCal then the error will cause the heater to increase it's duty cycle. But it's not exactly what we want! Lesser vUr means HOTTER cell. That's why we even have this safety check for overheating: (vUr < CJ125_UR_OVERHEAT_THR)...
So the simple trick is to inverse the error by swapping the target and input values.

martonmiklos, 2018-05-27:
Thanks for the clarification, I just opened this issue as a note.
If I see right the PID class is implemented as a generic class, so it might be useful to add a parameter for this inverted mode?

andreika-git, 2018-05-28:
> If I see right the PID class is implemented as a generic class, so it might be useful to add a parameter for this inverted mode?
In fact, the 'parameter' for inverted mode is already there. I mean, perhaps the better way is to call heaterPid.setErrorAmplification(-1.0f) somewhere in the code, and then we can just use normal mode: heaterPid.getValue(vUrCal, vUr) without those confusing swaps...

closed 2019-05-31 by rusefi