rusefi/rusefi #607: A few tables are Celsius-only not adjustable

rusefi, 2018-08-26:
On startup we select Fahrenheit or Celsius but some tables do not care

   curve = iatFuelCorrCurve, "Intake air temperature fuel Multiplier"
      columnLabel = "Coolant", "Multiplier"
      xAxis       =  -40, 120, 10

	curve = cltTimingCorrCurve, "Warmup timing correction"
      columnLabel = "Coolant", "Extra"
      xAxis       =  -40, 120, 10
      yAxis       =  0,  50, 10
      xBins       = cltTimingBins, coolant

	curve = cltFuelCorrCurve, "Warmup fuel manual Multiplier"
      columnLabel = "Coolant", "Multiplier"
      xAxis       =  -40, 120, 10
      yAxis       =  0,  50, 10
      xBins       = cltFuelCorrBins, coolant

	curve = crankingCltCurve, "Cranking Coolant Temperature multiplier"
      columnLabel = "Coolant", "Multiplier"
      xAxis       =  -40, 120, 10
      yAxis       =  0,  5, 10
      xBins       = crankingFuelBins, coolant

and probably more

See examples like
#if CELSIUS          
         filter = minCltFilter, "Minimum CLT", coolant,       <       , 60,       , true
#else
         filter = minCltFilter, "Minimum CLT", coolant,       <       , 160,      , true
#endif
and
#if CELSIUS
      coolant         = scalar,  F32,      4,              "Â°C",            1,       0.0
#else
      coolant         = scalar,  F32,      4,              "Â°F",         {9/5},  17.77777
#endif

for how things should be corrected

rusefi, 2018-08-27:
This would require a change to .ini generator - at the moment we have
float[IAT_CURVE_SIZE] iatFuelCorrBins;;"C",        1,     0,   -100.0,    250.0,   2

would need to teach generator that these are not just float but temperature floats so that an in-statement is generated

rusefillc, 2025-08-10:
dup https://github.com/rusefi/rusefi/issues/4788

PR #8375 (2025-08-01): migrate all temperature configs to "SPECIAL_CASE_TEMPERATURE" unit
Resolves: #607
needs: #8313

Migration of all tables/values/curves from "C" to "SPECIAL_CASE_TEMPERATURE", one per commit
closed 2025-08-10 by rusefillc