rusefi/rusefi #617: CJ125 unit-tests coverage

andreika-git, 2018-10-28:
The CJ125 controller code is potentially dangerous. It controls a powerful MOSFET and high-current WBO heater circuit. If any of this PID control logic fails, it may cause severe damage of WBO sensor or CJ125 hardware. 
Besides, the CJ125 chip is not that protected as it seems...
[image]

We're already started using our CJ125 code not only in the 'lab' conditions, but on the real cars. That's why we need to take some extra precautions when making any change to existing implementation...

mck1117, 2018-12-21:
FYI, I'm using the cj125 on my daily driver.  I've put ~1500 miles on it and it works great.

mck1117, 2023-02-20:
it looks like we have cj125 tests now?

closed 2023-02-20 by rusefillc