rusefi/rusefi #638: Enable multi core make by default

rusefi, 2018-12-28:
While Eclipse seems to be enabling multi-core, command line still compiles using only one core

Is there any chance to add any magic into the Makefile to compile say using 4 cores by default?

rusefi, 2018-12-28:
@fpoussin @mck1117 any takers?

rusefi, 2018-12-31:
https://stackoverflow.com/questions/43367269/why-makeflags-j12-is-not-working-anymore-with-make-4-2

mck1117, 2018-12-31:
Agree with @fpoussin, I'm not sure it's possible to do only with the makefile.  It's easily possible if you're building from the script or IDE, as those lets you edit the command that's run when it builds.

I misspoke the other day, it turns out it's possible in cmake, but maybe not possible from a regular makefile.

mck1117, 2020-07-28:
this is done

closed 2020-07-29 by rusefillc