rusefi/rusefi #640: Occasional halts on "Burn" in TunerStudio ("Red" LED is on)

andreika-git, 2018-12-29:
The engine must be running. Just change one of the settings and press "Burn" (may take a couple of attempts).
TunerStudio may complain about "CRC error" (not always).
Fuel pump stays enabled (no turn-off timeout).
Connected via TTL-USB bridge.

Caution! We must check the pin states in seg-fault mode to prevent a disaster (injectors stuck etc.)!

rusefi, 2018-12-29:
@andreika-git great minds think alike - we already seem to be turning pins off in case of FATAL

There are two scenarios of ON_FATAL_ERROR - there is chDbgPanic3 related to CH_CFG_SYSTEM_HALT_HOOK which I believe is used for segfault and there is firmwareError in case of software error. Blue LED would only be blinking in case of firmwareError, it would be great to know which case is your case.

```
/**
 * low-level function is used here to reduce stack usage
 */
#define ON_FATAL_ERROR() \
		palWritePad(LED_ERROR_PORT, LED_ERROR_PIN, 1); \
		turnAllPinsOff(); \
		enginePins.communicationLedPin.setValue(1);```

rusefi, 2019-01-05:
@andreika-git @darxfame is this still an issue? Were you able to reproduce this?

andreika-git, 2019-01-05:
We were able to reproduce it only with our experimental VSS-on-non-trigger-pins firmware, and it's related. So this is a specific issue, and EXT-INT version of VSS will definitely solve it.

andreika-git, 2019-01-11:
We've hunted down this bug!
The problem was in "registeredIcus", when driver==null. The function addWaveAnalyzerDriver() adds to this list for null driver, but removeWaveAnalyzerDriver() doesn't remove from it and exits prematurely.
I've attached the patch that solved the problem. I don't know if this is applicable to the official codebase or just fixes very rare case of our internal VSS-at-any-pin-patch...
[patch-removeWaveAnalyzerDriver().patch.zip](https://github.com/rusefi/rusefi/files/2751018/patch-removeWaveAnalyzerDriver.patch.zip)

rusefi, 2019-01-11:
Thank you for tracking the problem down, it's _extremely_ important to get to root causes if possible.

I believe the fix should be different - we should not even be adding NULL into the array

So what seems to happening - we were adding NULL and not removing and the array was overfilling and this was causing either a memory corruption or assertion error.

Open question how to fix. I believe array overfilling should stay a fatal issue, same as removal from empty. On the higher level code, we should not be adding NULL drivers into the array at all. 

Also, I would strongly suggest having EFI_ENABLE_ASSERTS

rusefi, 2019-01-11:
(did it just loose my comment?)

I strongly suggest running with EFI_ENABLE_ASSERTS. Great news that we were able to track root cause of this one!

What do you think about https://github.com/rusefi/rusefi/commit/d55c7dd856e574a9fea9d12c434d966a1e0748d6 ? Shall we close this ticket? Does anyone want to try the fix? Looks like the key thing was having invalid input pin and making enough burns so that the array would overfill

rusefi, 2019-01-11:
@darxfame @andreika-git FYI ^^^

andreika-git, 2019-01-11:
I'm happy with d55c7dd, and it's ok to close the ticket. This fix is not fully compatible with the old VSS patch (where any input pin works on additional timer), so it's better to wait until the new EXT_INT implementation is ready, and then re-test it.

closed 2019-01-11 by rusefi