rusefi/rusefi #643: Revive stm32f1 config

rusefi, 2019-01-03:
Back when we were compiling with Eclipse we had config/stm32f1egt configuration working. Looks like we have not migrated it to ChibiOS version we currently use

It would be needed to revive this configuration so that we can develop smaller CAN-based devices

rusefi, 2019-01-03:
@fpoussin would you be interested to look?

fpoussin, 2019-01-03:
F1 is filled with silicon bugs, uses old peripherals, and can't use USB with CAN...
STM32F0/F3/L4 are a much better choice depending on needs.

rusefi, 2019-01-03:
So F1 is not the luckiest product? They never managed to fix it?

Damn, eBay is so full of these sweet STM32F103C8T6 boards :( What would be a similar under $2 or under $5 board you would recommend?

fpoussin, 2019-01-03:
Nope, these are permanent silicon bugs.
For CAN devices we would not use breakout boards anyway, so F0 should be best in most cases.

rusefi, 2019-01-03:
No breakout boards for devices, but need breakout boards for prototypes. What is the cheapest useful F0 board?

fpoussin, 2019-01-03:
https://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-eval-tools/stm32-mcu-eval-tools/stm32-mcu-nucleo/nucleo-f042k6.html is a good option

mck1117, 2019-01-03:
Since ST has committed to not changing functionality within a family, those bugs are now a permanent part of any f1, future or current, forever.

rusefi, 2019-01-03:
I will get a couple of official nucleo-f0, but why are no $2 boards available from China? Is F0 more expensive than F1?

anyway, who wants to PR F0 config for nucleo?

fpoussin, 2019-01-05:
F0 is newer and less common than F1.
Make sure to get the Nucleo 042 and not 030.

mck1117, 2019-08-24:
can this issue be closed? f0 >>> f1

closed 2019-08-24 by rusefi