rusefi/rusefi #646: Looks like .hex files are not generated with gcc 8

rusefi, 2019-01-04:
[image]

rusefi, 2019-01-05:
full make output 
[image]

rusefi, 2019-01-05:
https://bugs.launchpad.net/gcc-arm-embedded/+bug/1810274

rusefi, 2019-01-05:
My version is gcc version 8.2.1 20181213 (release) [gcc-8-branch revision 267074] (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 

@andreika-git  what is your version?

[image]

andreika-git, 2019-01-05:
Mine is 8.2.1-2018.11 (gcc-arm-8.2-2018.11-i686-mingw32-arm-eabi.tar.xz)

andreika-git, 2019-01-05:
gcc -v output:
gcc version 8.2.1 20180802 (GNU Toolchain for the A-profile Architecture 8.2-2018.11 (arm-rel-8.26))

rusefi, 2019-07-14:
We have fresh build at https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

Version 8-2019-q3-update
Released: July 10, 2019

rusefi, 2019-07-14:
gcc 8.3 from 2019 build is fine

closed 2019-07-14 by rusefi