rusefi/rusefi #652: unit_tests should compile and run both on linux and cygwin windows

rusefi, 2019-01-07:
at the moment build server runs on cygwin windows and unit_tests compile and run there

@mck1117 reports that unit tests do not compile on unix, https://github.com/rusefi/rusefi/pull/651 was suggested

unfortunately while https://github.com/rusefi/rusefi/pull/651 fixes linux, it breaks cygwin :(

rusefi, 2019-01-07:
@mck1117 @fpoussin here is how windows fails with suggested change. Revering #651 for now

[image]

fpoussin, 2019-01-07:
[image]
Small tip for next time you need to revert a PR, just use the button.

fpoussin, 2019-01-07:
```
Compiling microsecond_timer.cpp
In file included from ../firmware/controllers/system/ThreadController.h:12:0,
                 from ../firmware/controllers/system/PeriodicController.h:10,
                 from ../firmware/hw_layer/microsecond_timer.cpp:18:
../firmware/ChibiOS4/os/rt/include/ch.h:70:10: fatal error: chtypes.h: No such file or directory
 #include "chtypes.h"
          ^~~~~~~~~~~
compilation terminated.
rules.mk:100: recipe for target 'build/obj/microsecond_timer.o' failed
make: *** [build/obj/microsecond_timer.o] Error 1
```
With gcc 7.3.0

rusefi, 2019-01-07:
We have two defects on top of each other. At the moment things do not compile abywhere due to recent code change, but usually they do not compile on linux only

fpoussin, 2019-01-07:
Ok, let me know when it's fixed.

rusefi, 2019-01-07:
Things now compile on windows. How is linux?

fpoussin, 2019-01-07:
Better, now there's only the pthread linker issue.

fpoussin, 2019-01-07:
Which version of mingw are you using?

rusefi, 2019-01-07:
Fresh cygwin gives me 7.3 or 7.4, for instance on this device 
[image]

commit 5b5f4e12 (2019-01-09): good enough fix #652
closed 2019-01-09 by rusefi