rusefi/rusefi #653: unit tests code coverage

rusefi, 2019-01-09:
maybe gcov?

1) first step building with Makefile changes => .gcno "gcov notes" files are generated

rusefi, 2019-08-25:
https://stackoverflow.com/questions/57643813/cannot-get-gcov-to-produce-any-coverage-data

rusefi, 2019-08-25:
Pile of BS :(
With a HelloWorld.cpp application I've realized that x86_64-w64-mingw32-g++ and i686-w64-mingw32-g++.exe generated main.exe does not produce main.gcda

while plain g++ does produce main.gcda!

but plain g++ does not compile rusEfi because
```googletest/googletest/include/gtest/internal/gtest-port.h:2496:40: error: 'fileno' was not declared in this scope```

https://stackoverflow.com/questions/18784112/googletest-1-6-with-cygwin-1-7-compile-error-fileno-was-not-declared-in-this
helps to compile but now I am having linker issues
```
Linking build/rusefi_test
/usr/lib/gcc/x86_64-pc-cygwin/7.4.0/../../../../x86_64-pc-cygwin/bin/ld: symbol `__stack_chk_guard' missing from main hash table
/usr/lib/gcc/x86_64-pc-cygwin/7.4.0/../../../../x86_64-pc-cygwin/bin/ld: symbol `environ' missing from main hash table
```

rusefi, 2019-08-25:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=854368 but they say closed?

rusefi, 2019-08-25:
```
# mingw-w64 is a project which forked from mingw in 2007.
# It looks like cygwin build of mingwg-w64 has issues with gcov runtime :(
# In order to have coverage generated please download from https://mingw-w64.org/doku.php/download/mingw-builds
# Install using mingw-w64-install.exe instead of similar thing packaged with cygwin
```

chuckwagoncomputing, 2020-08-14:
Is there any point to keeping this issue around? Are we not set on keeping mingw as a requirement?

closed 2020-08-14 by rusefillc