rusefi/rusefi #657: The Big Refactoring of 2019: DECLARE_ENGINE_PARAMETER_SUFFIX consistency and simplification

rusefi, 2019-01-10:
DECLARE_ENGINE_PARAMETER_SUFFIX could be improved a bit easily

mck1117, 2019-01-11:
I think before we jump in and rework the use of the ENGINE_PARAMETER_SUFFIX, we need to look at whether it's the right way to pass configuration and state around the workings of the ECU (hint, it isn't).  Let's not waste our limited time on something we're going to change later anyway.

rusefi, 2019-01-16:
See also https://github.com/mck1117/efi-playground/blob/master/configurations/tester/main.cpp

See also https://github.com/mck1117/rusefi/tree/mck1117/cj125-pr-3/firmware/hw_layer/sensors

rusefi, 2019-01-28:
https://gustedt.wordpress.com/2010/06/08/detect-empty-macro-arguments/

closed 2024-02-26 by rusefillc