rusefi/rusefi #658: Consult: VVT mode and phase sensor[Rus]

darxfame, 2019-01-10:
У нас в прошивке есть текущая реализация метода VVT в режиме firt half и second half, так же в прошивке есть несколько Operation mode для триггеров, а именно меня интересуют 4 stroke with cam sensor и  4 stroke without cam sensor.
Так вот. У меня автомобиль с конфигурацией основное зубчатое колесо на коленчатом валу 60-2
Так же присутствует датчик распределительного вала, который имеет 1 зуб и он подает сигнал на ЭБУ за 20 градусов до ВМТ

Так вот, я хотел бы узнать, для чего используется VVT режимы, как они работают если стоит 4 stroke without cam sensor, используется ли при этом как то прошивкой датчик фазы или нет и по сути является ли VVT настройкой для датчика фаз.

И если датчик фаз используется в режиме 4 stroke without cam sensor то для чего нужен режим 4 stroke with cam sensor.

rusefi, 2019-01-10:
@darxfame я тебя не очень понимаю :(

Пожалуйста, пронумеруй вопросы. В конце каждого вопроса пожалуйста поставь вопросительный знак. Я не придираюсь и не издеваюсь - мне правда нужна ясность.

darxfame, 2019-01-10:
> @darxfame я тебя не очень понимаю :(
> 
> Пожалуйста, пронумеруй вопросы. В конце каждого вопроса пожалуйста поставь вопросительный знак. Я не придираюсь и не издеваюсь - мне правда нужна ясность.

1. Чем различаются режимы 4 stroke with cam sensor и 4 stroke without cam sensor для триггера 60-2?
2. Что такое режимы VVT?
3. Почему в консоли явно не указан датчик фаз (я так понимаю он является частным случаем VVT, но все таки это вводит в заблуждение)?
4. Почему нет режима где датчик фаз отсутствует в принципе, т.е VVT mode **OFF**?
5. Почему включение и отключение режима  VVT определяется только наличием пина в настройке VVT Sensor input?
6. Почему нет поля указания фазы впрыска на пуске и почему она берется откуда то из основной таблицы?

rusefi, 2019-01-10:
@darxfame 

Вход распредвала может использоваться либо просто для определения фазы двигателя, если ваша форма триггера не включает датчик распредвала в качестве основного канала, либо он может использоваться для положения распредвала в случае системы изменяемых фаз.

Camshaft input could be used either just for engine phase detection if your trigger shape does not include cam sensor as 'primary' channel, or it could be used for Variable Valve timing on one of the camshafts.

darxfame, 2019-01-10:
> @darxfame
> 
> Вход распредвала может использоваться либо просто для определения фазы двигателя, если ваша форма триггера не включает датчик распредвала в качестве основного канала, либо он может использоваться для положения распредвала в случае системы изменяемых фаз.
> 
> Ещё у нас есть ссылка на https://rusefi.com/wiki/index.php?title=Manual:Software:VVT
> 
> Добавил в исходнки комментарий
> Camshaft input could be used either just for engine phase detection if your trigger shape does not include cam sensor as 'primary' channel, or it could be used for Variable Valve timing on one of the camshafts.

Хорошо, первые 2 вопроса отпали, идем дальше

rusefi, 2019-01-10:
3. В студии надо как-то все эти вопросы прояснить, сейчас там плохо. В исходниках проекта студии есть 
	    field = "!https://rusefi.com/wiki/index.php?title=Manual:Software:VVT"
но я почему-то не вижу этой ссылки нигде?

это всё сложные темы, у меня не получается к ним сделать хороший интерфейс и хорошее понятное описание, мне нужна в этом помощь.

darxfame, 2019-01-10:
> 1. В студии надо как-то все эти вопросы прояснить, сейчас там плохо. В исходниках проекта студии есть
>    field = "!https://rusefi.com/wiki/index.php?title=Manual:Software:VVT"
>    но я почему-то не вижу этой ссылки нигде?
> 
> это всё сложные темы, у меня не получается к ним сделать хороший интерфейс и хорошее понятное описание, мне нужна в этом помощь.

Тогда по поводу 3 пункта я предлагаю выделить обычный ДФ в отдельный блок в Trigger Configuration и использовать отдельно либо Cam Sensor либо VVT Cam Sensor

darxfame, 2019-01-10:
> 1. В студии надо как-то все эти вопросы прояснить, сейчас там плохо. В исходниках проекта студии есть
>    field = "!https://rusefi.com/wiki/index.php?title=Manual:Software:VVT"
>    но я почему-то не [...]

А вот та ссылка, что ты указал выше
[image]

rusefi, 2019-01-10:
4. До конверсии проекта студии шаблон был "нет входного пина - нет фичи, пины тогда обычно были рядом с параметрами. Сейчас после конверсии проекта нужно про это заново подумать, как сделать хорошо.

а про ссылку - ну вот какой я слепой :)

darxfame, 2019-01-10:
> 1. До конверсии проекта студии шаблон был "нет входного пина - нет фичи, пины тогда обычно были рядом с параметрами. Сейчас после конверсии проекта нужно про это заново подумать, как сделать хорошо.
> 
> а про ссылку - ну вот какой я слепой :)

Хорошо, предположим 4 и 5 пункт мы закрыли, остался 6-й

mck1117, 2019-01-10:
Is it unreasonable to ask for English-only on GitHub issues?

darxfame, 2019-01-10:
> Is it unreasonable to ask for English-only on GitHub issues?

unfortunately i dont know english well for this....

rusefi, 2019-01-11:
6. Почему нет поля указания фазы впрыска на пуске и почему она берется откуда то из основной таблицы?

я не понимаю вопрос

darxfame, 2019-01-11:
@rusefi у нас есть таблица, называется фаза впрыска, но она не предусматривает малые обороты двигателя.
Поэтому для режима cranking используется какое то значение из этой таблицы, что не правильно, потому как мы должны использовать или отдельную таблицу для cranking injection phase или отдельный field в котором мы укажем одно значение injection phase для режима cranking

rusefi, 2019-01-11:
Ага, fuelPhaseMap и injectionPhase - это мы совсем от оригинальной темы удалились

не понимаю - почему она не предусматривает малые обороты двигателя? что мешает при желании поставить любые шкалы снизу и поставить одинаковое значение для всех давлений ниже 500 оборовто?

darxfame, 2019-01-11:
> Ага, fuelPhaseMap и injectionPhase - это мы совсем от оригинальной темы удалились
> 
> не понимаю - почему она не предусматривает малые обороты двигателя? что мешает при желании поставить любые шкалы снизу и поставить одинаковое значение для всех давлений ниже 500 оборовто?

Ну потому что это костыль. Так же как то что у нас включение датчика фаз сделано в vvt и через пин

rusefi, 2019-01-11:
да, это в сторону костыля. с другой сторойны, насколько мы уверены, что это вообще кому-либо важно - фаза впрыска на старте? это явно отдельный тикет, это очень далеко от наших триггерных забот здесь

darxfame, 2019-01-11:
> да, это в сторону костыля. с другой сторойны, насколько мы уверены, что это вообще кому-либо важно - фаза впрыска на старте? это явно отдельный тикет, это очень далеко от наших триггерных забот здесь

Можно не заводить отдельный тикет, а сделать под каким нибудь

rusefi, 2019-01-11:
давай заведём отдельный, чтоб была запись-мечта. у этого тикета про crankingInjectionPhase нет шансов, что я буду его делать скоро :(

rusefi, 2019-01-14:
@darxfame закрываем видимо этот тикет? ясность появилась?

darxfame, 2019-01-14:
@rusefi в принципе да, с пометкой решить задачи под тикетом.

rusefi, 2019-01-14:
@darxfame закрываю этот тикет - если хочешь, заводи тикет третьего пункта, или можно забить здесь и тикеты не плодить, раз совсем нет вероятности прогресса :(

closed 2019-01-14 by rusefi