rusefi/rusefi #659: BUG: phase sensor validation attending

darxfame, 2019-01-10:
В текущей реализации из моей практики прошивка не определяет наличие сигнала датчика фазы.
Так же не выводится в консоли (на сколько я видел) информация о том что датчик отсутствует или имеет неверный сигнал.
Сейчас человек, у которого произошел обрыв датчика фаз имеет шансы 50 на 50 запустить автомобиль, потому как прошивка случайным образом определяет фазу впрыска при включении зажигания...

Правильно:
- Информировать человека об отсутствии датчика фаз в консоли
- Выдавать сигнал на лампу Check Engine
- Запоминать ошибку отсутствия данного датчика
- Включать аварийный - попарный режим впрыска

English translate with Google
In the current implementation from my practice, the firmware does not detect the presence of a phase sensor signal.
It also does not display in the console (as far as I saw) the information that the sensor is missing or has the wrong signal.
Now a person who has a phase sensor break has a chance to start the car 50 to 50, because the firmware randomly determines the injection phase when the ignition is turned on ...

Right:
- Inform the person about the absence of a phase sensor in the console
- To issue a signal to the Check Engine lamp
- Remember the error of the absence of this sensor
- Enable emergency - pairwise injection mode

rusefi, 2019-05-11:
See also #669

rusefi, 2019-05-11:
Implemented using new 'vvtCamCounter' variable on TriggerCentral / TriggerState

See https://github.com/rusefi/rusefi/commit/812c61903ae5fd005232344c329c9d7f043fe64b

commit 812c6190 (2019-05-11): BUG: phase sensor validation attending - detect missing CAM signal fix #659
closed 2019-05-11 by rusefi