rusefi/rusefi #661: Usability: Save error codes in NVRAM?

andreika-git, 2019-01-11:
Here's an idea to store "last warning/error codes" in NVRAM (backupRamSave()/Load()). Last one or two is enough, I think (2x16bit?). Currently we have only lastErrorCode which resets every time when the firmware starts, but we might also add savedErrorCode(s), and load them at the start (and update when an error occurs). We will also need a command to reset (erase) the stored error codes. 
I think using NVRAM is more safe & fast than storing it in Flash (as normal settings), and allows to save it literally from anywhere in the code (even seg.fault handler?)

rusefi, 2019-01-11:
I like this idea! Any chance for a PR? :)

I would make it more than just one or two. We actually also need a way to display a larger number of error codes in TS, maybe using debug info mode? I believe we should be remembering 4 or 8 last error codes.

rusefi, 2019-01-11:
see also https://github.com/rusefi/rusefi/commit/aeea2b95d090c72669f695fe114493b27aab54b7

rusefillc, 2025-07-19:
wow we actually have this now!

closed 2025-07-19 by rusefillc