rusefi/rusefi #662: Question: something strange trigger errors

darxfame, 2019-01-11:
In my last trip i have got a lot of trigger errors with strange values. I attach my log file into this issue to parce it.
In the bottom this post you can see timing in log file and description a trigger errors in this timing.

1) 110s: CUSTOM_OBD_SKIPPED_FUEL
2) 168s: CUSTOM_ERR_6548 ([RUS]Внимание! Встречается два раза в разных местах - getAdvance() и endAveraging() - так быть не должно) ([ENG]Attention! It occurs twice in different places - getAdvance () and endAveraging () - this should not be so)
3) 273s: CUSTOM_RE_ADDING
4) 470s: CUSTOM_OUT_OF_ORDER_COIL
5) 471s: CUSTOM_TOO_LONG_FUEL_INJECTION

rusefi, 2019-01-11:
Thank you for the data, this is valuable information! This requires further investigation and I will try to spend some time on this in the coming days, I absolutely agree that this is really important.

I will try to reproduce any of these conditions with a unit test and we will go from there.

andreika-git, 2019-01-11:
@darxfame asked me to attach his log-file:

[2018-12-01_23.11.47.msl.zip](https://github.com/rusefi/rusefi/files/2750244/2018-12-01_23.11.47.msl.zip)

darxfame, 2019-01-11:
thanks @andreika-git

rusefi, 2019-01-13:
Just a status update: some unit tests are already reproducing at least some of the error codes.

Next step would be for me to revisit these unit tests and document the conditions a bit better and we can discuss what we do next. But that's already a lot - we have code which reproduces your condition!

https://github.com/rusefi/rusefi/commit/85a4307d044a7963cc8a615686704a2a9643acfb

rusefi, 2019-01-14:
See also #668

andreika-git, 2019-01-14:
We should definitely add error 6505 to the list (CUSTOM_ERR_START_STACK). It appears at the very start of the firmware, long before cranking... The above log is not complete (no cranking stage recorded), but the error 6505 exists in all other full logs (both mine and darxfame's).

rusefi, 2019-01-14:
interesting, let me fix that
```
	if (cisnan(coolantTemp) || cisnan(airTemp)) {
		warning(CUSTOM_ERR_START_STACK, "t-getTCharge NaN");
		return coolantTemp;
	}
```

rusefi, 2019-01-14:
I hope that https://github.com/rusefi/rusefi/commit/4b72599e2fe91c546a5a0504dce23f353b7be42c fixes CUSTOM_ERR_START_STACK

rusefi, 2019-01-16:
@andreika-git @darxfame what are open issues on this ticket?

darxfame, 2019-01-16:
Forgive me, but i not check this issuse a few days. I hope to read all in this issue at the this week's holiday

OrchardPerformance, 2020-05-31:
@darxfame - Any update?