rusefi/rusefi #667: BUG: fatal in firmware with DEBUG mode

darxfame, 2019-01-11:
I compile current firmware with flag `set DEBUG_LEVEL_OPT="-O2"` and firmware starts normally,
But if i compile it without this flag, i've got a `WARNING: HW timer state 1` message and FATAL firmware status. red and blue LED is always ON

Log file from rusefi console
[MAIN_rfi_report_2019-01-11_22_00_37.zip](https://github.com/rusefi/rusefi/files/2751039/MAIN_rfi_report_2019-01-11_22_00_37.zip)

darxfame, 2019-01-11:
`cd rusefi/firmware
set PROJECT_BOARD=Prometheus
set EXTRA_PARAMS="-DDUMMY -DSTM32F469xx -DEFI_ENABLE_ASSERTS=FALSE -DCH_DBG_ENABLE_CHECKS=FALSE -DCH_DBG_ENABLE_TRACE=FALSE -DCH_DBG_ENABLE_ASSERTS=FALSE -DCH_DBG_ENABLE_STACK_CHECK=FALSE -DCH_DBG_FILL_THREADS=FALSE -DCH_DBG_THREADS_PROFILING=FALSE"
set DEBUG_LEVEL_OPT="-O2"
set USE_BOOTLOADER=yes
call compile_and_program.bat -r -j4
`

rusefi, 2019-01-11:
This command does not work :( I suspect that's not a real command? It could not be just one huge line

maybe you need to use
```
three tilde syntax - start with three tilde and end with three tilde
```
 

I believe it should be possible to produce a real command - maybe take compile_and_program.bat for start? Maybe ";" are missing?

[image]

rusefi, 2019-01-11:
maybe attach a .bat file?

darxfame, 2019-01-11:
> может прикрепить файл .bat?
@rusefi attach it
[!compile-prometheus.zip](https://github.com/rusefi/rusefi/files/2751364/compile-prometheus.zip)

darxfame, 2019-01-12:
[arm-none-eabi-gcc-v.txt](https://github.com/rusefi/rusefi/files/2751845/arm-none-eabi-gcc-v.txt)

andreika-git, 2019-01-13:
Forgot to mention that is was compiled with USE_FPU = softfp

andreika-git, 2019-01-14:
Also I have to remind that earlier firmware versions worked fine with DEBUG/RELEASE mode on the same config. We need to investigate when it happened - maybe during ChibiOS3->4 migration?

rusefi, 2019-01-16:
@andreika-git do you have hardware and interest to look into this? There is no change that I will ever have time go grab my F469 and investigate :( maybe @fpoussin has a F469

andreika-git, 2019-01-16:
I do have hardware and interest, but I have no human resources currently :( Maybe I'll find some time in two weeks...

mck1117, 2021-02-13:
seems like it's been a long time since this, bug probably fixed?

closed 2021-02-13 by rusefillc