rusefi/rusefi #669: warning code on normal engine start up

rusefi, 2019-01-14:
It is reported that sometimes (often?) normal engine start-up gives you a warning code. We should not have a warning code on normal engine start-up

rusefi, 2019-01-14:
@andreika-git @darxfame which code was it that by the way?

TODO: would be nice to reproduce issue with a unit test first

andreika-git, 2019-01-14:
Mostly it's CUSTOM_SYNC_COUNT_MISMATCH.
[start-err.zip](https://github.com/rusefi/rusefi/files/2754123/start-err.zip)

rusefi, 2019-05-11:
Some progress was just made, need to test again

```
			// We only care about trigger shape once we have synchronized trigger. Anything could happen
			// during first revolution and it's fine
```

rusefi, 2019-05-11:
Open question: SD card warning should it be a warning only once? not a warning at all even if SD card was activated? an option to show or not show warning on SD card issue maybe?

rusefi, 2019-05-11:
[image]

[image]

another source of warning: IAT and CLT are calculated before first slow ADC result

rusefi, 2019-05-11:
Above issue fixed, but still trigger cranking warnings on real Miata NA6

[DataLogs_cranking_produces_warnings.zip](https://github.com/rusefi/rusefi/files/3169181/DataLogs_cranking_produces_warnings.zip)

rusefi, 2019-05-11:
Still a bunch of trigger warnings of real Miata NA
[crankingNA_logic_data.zip](https://github.com/rusefi/rusefi/files/3169277/crankingNA_logic_data.zip)

rusefillc, 2021-05-22:
[image]

[image]

rusefillc, 2021-06-25:
Relates to #2855