rusefi/rusefi #670: testFasterEngineSpinningUp unit test does not run on it's own

rusefi, 2019-01-14:
unit tests should not depend on each other, each unit test should perform exactly the same regardless of test order

At the moment it looks like "testFasterEngineSpinningUp" depends on order of execution, this test does not want to run on it's own. Looks like we have some state persisted between unit tests, that's probably a wider problem not exactly specific to testFasterEngineSpinningUp

Steps to reproduce:

in main.cpp go to "uncomment if you only want to run selected tests" line and make it
```
	// uncomment if you only want to run selected tests
	::testing::GTEST_FLAG(filter) = "*testFasterEngineSpinningUp*";
	return RUN_ALL_TESTS();
```

This would tell googletest to only run one test

andreika-git, 2019-01-14:
#673 Fixed

rusefi, 2019-01-16:
This one is resolved but we have discovered a different issue #674

closed 2019-01-16 by rusefi