rusefi/rusefi #683: SD card in SDIO mode

rusefi, 2019-01-26:
Random boards like https://wiki.stm32duino.com/images/6/69/Vcc-gnd.com-STM32F407VET_mini-schematic.pdf have SD card wired to SDIO. While SDIO shares pins with SPI3, I believe SPI code would not work for instance because SDIO clock is PC12 but SPI clock is PC10.

We need SDIO SD card implementation.

ahmadmn, 2019-09-08:
is it useful example:
[](http://stm32f4-discovery.net/2014/07/library-21-read-sd-card-fatfs-stm32f4xx-devices/)

rusefillc, 2021-01-05:
@dron0gus @andreika-git we might need faster SD write for "smart ecu" meta epic. Open question if SDIO would help with performance or not.

closed 2021-02-25 by rusefillc

rusefillc, 2021-05-04:
We now have SDIO mode as compile time option EFI_SDC_DEVICE

No official boards are using SDIO as of today