rusefi/rusefi #692: USB issue F767

960, 2019-02-11:
After the latest updates something happened to the USB connection for F767.

It assigned another USB port(Had to reinstall the VCP), and new port number.

Nothing works anymore, and it just freezes in tunerstudio.

960, 2019-02-13:
Actually it boils the CPU. 
Have tried another, and same issue.

Compiled with version before ChiBios update, old USB port is back and things are normal.

rusefi, 2019-02-13:
@960 please define "boils the CPU"
was the CPU hot or is it now permanently damaged?

For the next week I do not have any F7 devices to test with. Feel free to ship me an F767 since I do not have one at all.

In 9 days from today I hope to try on my F746

960, 2019-02-13:
It's damaged, two of them.

I dont know whats causing it. 
I noticed every time I tried connecting to tunerstudio it only took some seconds before it "froze". There was no valid settings, and could not load Tunerstudio-settings. A new VCP was assigned for both.

rusefi, 2019-02-13:
@960 I am not 100% sure we are not 100% on the same page with definition of "damaged"

Can you please compile chibios F769 demo and flash it to confirm it would blink status LED? "damaged" for me means "fucked up beyong repair, time to throw it away into the trash". I would be surprised if that's the case

rusefi, 2019-02-13:
Also new VCP is expected, that's probably a change to USB unique ID between CHibiOS 17 and ChibiOS18

please also try "erase chip" to reset configuration to complete blank state using ST Utility

rusefi, 2019-02-21:
Can you reproduce the issue with https://rusefi.com/build_server/rusefi_bundle_stm32f767.zip pre-compiled binaries?

rusefi, 2019-02-22:
@960 ?

960, 2019-03-15:
Looks good on Nucleo, closing issue now.

closed 2019-03-15 by 960