rusefi/rusefi #696: fail to generate F7 DFU files

rusefi, 2019-02-23:
```Loading build/rusefi.hex...
ERROR: Image appears to have invalid initial program counter```

:( See https://community.st.com/s/question/0D50X00009XkeovSAB/dfu-fail-verify-8-byte-align-magic

fpoussin, 2019-02-25:
Need more context. Which program gives this error?

rusefi, 2019-02-25:
the command in build script:
..\misc\hex2dfu\HEX2DFU.exe build/rusefi.hex -out build/rusefi.dfu

this is an utility from Clive 2.0 see https://community.st.com/s/question/0D50X00009XkeovSAB/dfu-fail-verify-8-byte-align-magic

@andreika-git has just implemented an alternative utility also there is something on https://github.com/encedo/hex2dfu

rusefi, 2019-06-08:
https://github.com/encedo/hex2dfu needed some changes to compile so we now have https://github.com/rusefi/hex2dfu

closed 2019-07-30 by rusefi