rusefi/rusefi #702: Pre-defined triggerOffsets for OEM trigger wheels

andreika-git, 2019-03-11:
It seems that a precise triggerOffset definition is non-trivial even for reasonably advanced users.
That's why I suggest to prepare some pre-defined values for popular OEM trigger wheel configurations like 60-2. And set them as defaults when the trigger shape is changed.

The problem is that "common sense" teeth counting is non-intuitive (0-based trigger event counting logic is used in the firmware), and 20-th tooth is not 20*6=120 degrees...

rusefi, 2019-03-11:
@andreika-git I believe you are proposing to add "60/2 at 20th" trigger?
what is tdcOffset for such standard trigger?

andreika-git, 2019-03-11:
I believe it's 114:
[image]

OrchardPerformance, 2020-06-01:
Unfortunately the 60-2 has multiple variations, volvo is 90 degrees, early BMW is 84 degrees and the late BMW is 324 degrees. 

Best we can possibly do is to compile a list of as many as we can on the wiki.

mck1117, 2021-11-29:
we do this for very well known triggers

closed 2021-11-29 by mck1117