rusefi/rusefi #705: Serial-over-usb driver connectivity issue

andreika-git, 2019-03-14:
The existing Serial-over-USB driver (usbconsole.c) doesn't work on some STM32F407VET6 boards (See https://rusefi.com/forum/viewtopic.php?f=4&t=1498). The PC shows "usb device not recognized" error. 
It may depend on the PC hardware/OS and/or STM32 board/USB-cable/SWD-programmer connected or not, etc...
We suspect that the problem is with the following code:
`/*
	 * Activates the USB driver and then the USB bus pull-up on D+.
	 * Note, a delay is inserted in order to not have to disconnect the cable
	 * after a reset.
	 */
	usbDisconnectBus(serusbcfg.usbp);
	chThdSleepMilliseconds(1500);
	usbStart(serusbcfg.usbp, &usbcfg);
	usbConnectBus(serusbcfg.usbp);
`

rusefi, 2019-03-14:
I have five boards work fine on my Windows 7 64 bit using the following driver

[image]

what is your OS exactly? can you please attach similar driver info picture?

andreika-git, 2019-03-14:
The same Win7-64,
[image]

rusefi, 2019-03-14:
[rusefi_master_no_changes_mini_board.zip](https://github.com/rusefi/rusefi/files/2966594/rusefi_master_no_changes_mini_board.zip)

andreika-git, 2019-03-15:
http://www.chibios.com/forum/viewtopic.php?f=16&t=4981
(wasn't updated lately but still related...)

rusefi, 2019-05-11:
We have no idea what this issue is about, right?

mck1117, 2021-02-13:
USB serial is very stable these days, close?

closed 2021-02-13 by rusefillc