rusefi/rusefi #707: "60/2 with variable cam location" trigger

rusefi, 2019-03-15:
In order to avoid massive confusion about rusEfi trigger we need a way to define 60/2 + cam or 36/1 + cam as a trigger with variable cam shaft mark position, not as a VVT configuration

rusefi, 2019-03-15:
JRD McLAREN wrote: ↑Thu Mar 14, 2019 2:46 pm
I'm think, the right name for your trigger wheel templates are:

60/2 with Hall sensor (60/2)
60/2 with VR sensor (60/2 VW)

...some OPEL / Vauxhall car have HALL sensor on crank ...

See https://rusefi.com/forum/viewtopic.php?f=2&t=1519&p=32088#p32088

rusefi, 2019-04-29:
@andreika-git @darxfame @dron0gus we need to get back to this...

rusefi, 2019-04-29:
https://rusefi.slack.com/messages/CJ9JH3Y21/details/

I believe there was a table on github somewhere, need to check my skype history for the link

andreika-git, 2019-04-30:
https://github.com/rusefi/rusefi_documentation/blob/master/triggers/readme.md

mck1117, 2021-07-08:
this works now - you can configure a cam with "first half" or "second half" and things work properly

closed 2021-07-08 by mck1117