rusefi/rusefi #708: rusEfi F767 does not start on my custom board

960, 2019-03-15:
I have my own board design which seems to copy nuckeo144 in terms of MCU and everything around, yet F767 code does not start on F767 chip

same board everything works on F407 chip if I solder F407 chip

960, 2019-03-15:
SWD debugging

[image]

960, 2019-03-15:
http://www.chibios.com/forum/viewtopic.php?t=3274

960, 2019-03-15:
[image]

960, 2019-03-15:
I have now added 8Mhz. By the way I am on a newer edition of my board - I moved 8MHz closer to mcu, as it was totally wrong at the other. and changed to micro sized

stil does not work as is.

Since it's hanging near STM32_HSE_BYPASS

@rusefi recommended
```
//#define STM32_HSE_BYPASS
```
and it works like that - I got TunerStudio to connect via USB serial port!!!

What is this STM32_HSE_BYPASS and while disabling it helps?!

andreika-git, 2019-03-16:
Don't forget that Nucleo code didn't use the external crystal, and STM32_HSE_BYPASS is used for external clock to ignore (bypass) the crystal oscillator:

> Bit 18 HSEBYP: HSE crystal oscillator bypass
Set and cleared by software to bypass the oscillator with an external clock. The external clock
must be enabled with the HSEON bit set, to be used by the device. The HSEBYP bit can be
written only if the HSE oscillator is disabled.
0: HSE crystal oscillator not bypassed
1: HSE crystal oscillator bypassed with external clock

andreika-git, 2019-03-16:
So if you're using external **generator**, then you'll need to set STM32_HSE_BYPASS. But if you're using external quartz **crystal oscillator**, then disable it.

960, 2019-03-16:
What exactly are generating clock signal at the Nucleo?
I have tried with and without crystals in every combination.

I also have a Nucleo that I stripped down to a bare minimum. 
I removed both the 32.xxx and 25.xxx and it still ran just as fine.
The only thing I did not try was to break off the ST-Link. If it has a external source, it must come from there, and should not be enabled.

Has anyone tried to break off the ST-Link?

andreika-git, 2019-03-16:
Here's MCU side with OSC_IN, you can see the MCO signal there:
[image]
And here's the ST-Link side, you can see the MCO signal source there:
[image]

P.S. I have't tried to break the SL-Link off, because the Nucleo won't work as it is, I guess...

960, 2019-03-16:
I just saw it after finding the explanation at page 24 in this: https://www.st.com/content/ccc/resource/technical/document/application_note/b0/5c/ba/b2/cb/10/41/dd/DM00164549.pdf/files/DM00164549.pdf/jcr:content/translations/en.DM00164549.pdf

But this means HSE_BYPASS should not be set, as it should work without the ST-Link?

andreika-git, 2019-03-16:
If you want to make it work without ST-Link, then yes, you should not set HSE_BYPASS, and you should solder the X3 crystal instead, and caps (C38,C38) and also short the jumpers (SB8,SB9). Or use the internal oscillator, as an option...

960, 2019-03-16:
The internal would be the most correct in this case as it should work if anyone would remove the ST-Link to save space.
Will it work the same with the internal?

andreika-git, 2019-03-16:
For internal osc, you should change the STM32_PLLSRC_HSE into STM32_PLLSRC_HSI, I guess, and check other corresponding settings like STM32_HSI_ENABLED...

rusefi, 2019-03-16:
But I assume there is some value in external osc? Why is everyone using external osc instead of using internal osc to reduce complexity and costs?

andreika-git, 2019-03-16:
I guess the crystal (quartz) oscillator has better accuracy (less frequency errors) and temperature stability than the internal RC-oscillator. The quartz resonates only on certain frequencies (or their harmonics) based on the physical crystal size, and RC-circuit is just a plain frequency generator which can drift. 
Our firmware uses some precision timings (like in ignition spark scheduler) so it's rather important to have a good oscillator...

andreika-git, 2019-03-16:
The ST-Link chip and the Nucleo chip are syncronized by the same MCO osc. signal by default. And this is for a reason: it minimizes a phase jitter in high-speed digital connection between the chips. If each STM32 had a separate precise crystal oscillator, their frequencies would not be absolutely the same, and a small "delta-frequency" deviation would affect the phase of received signals causing a little unstability.

960, 2019-03-16:
The STM32_PLLSRC_HSI and STM32_HSI_ENABLED was in the ChiBios folders, so I dont know what to do with them.

Can you please check the files in my folder and add them? If you can add a "engine" config as well would be fine.

https://github.com/960/rusefi

rusefi, 2019-03-17:
I need to adjust build server to start building both versions of F767 - with and without the bypass
or I can change the F767 build and it would require that nucleo boards are modified

rusefi, 2019-03-20:
https://rusefi.com/build_server/ now has two versions:

!compile-f767.bat produces rusefi_bundle_stm32f767_osc.zip for when you have a real oscillator

and rusefi_bundle_stm32f767_nucleo.zip produces rusefi_bundle_stm32f767_nucleo.zip for when you use Nucleo with MCO

closed 2019-03-20 by 960