rusefi/rusefi #711: Watchdog-related firmware restart issue on STM32F405

andreika-git, 2019-03-21:
The rusEFI firmware compiled for Prometheus-405 doesn't run normally on STM32F405 (See https://rusefi.com/forum/viewtopic.php?f=4&t=1506)
The CPU is in permanent reboot mode (non-dependent on the code/address).
Current solution is to disable the hardware watchdog via flash option-bytes:
[image]

It seems that this setting is persistent and not affected by 'normal' firmware updates, so it can be set once...
Keywords: WDG_HW, WDG_SW, ...
More info needed...

mck1117, 2021-06-26:
@andreika-git Was this due to a previous setting of that bit manually? I've never seen this on either an F407 or F405 (which should be the same in this regard)

mck1117, 2021-06-26:
factory configuration on STM32 is with WDG_SW = true (as shown in the screenshot above)

andreika-git, 2021-06-27:
> @andreika-git Was this due to a previous setting of that bit manually? I've never seen this on either an F407 or F405 (which should be the same in this regard)

No, it was the problem with a new chip. I don't remember how we fixed it. Maybe it was a faulty chip and we replaced it, or we changed the firmware...